keirf/flashfloppy #885: 40 track HFE files result in error 31

sdsnatcher73, 2024-03-27:
So I have another semi obscure system, a Fujitsu FM-77L4 (a pre AV machine). This machine and early FM77AV machines have 40 track 3.5" DD disk drives. The drives used are actually YE-DATA YD625, [datasheet](http://www.bitsavers.org/pdf/yeData/FDK-523002_YD-600_Specifications_Jan85.pdf) is available. So 67.5TPI as opposed to 135TPI. All of course done to maintain full compatibility to 40 track 5.25" drives and media used on FM8 and FM7. Later models actually switched to 80 track drives that were made to double stap on media formatted as 40 track. But most software is in the 40 track format.

Images of software for these systems is typically distributed in D88 format (although files often have D77 extension). Here is an example of a disk image in D88 format:
[1942.d88.zip](https://github.com/keirf/flashfloppy/files/14780031/1942.d88.zip)

I have one of the 40 track AV systems with a Lotharek HxC SD model F. For this machine I converted the D88 image to a 40 track HFE file:
[1942.hfe.zip](https://github.com/keirf/flashfloppy/files/14780043/1942.hfe.zip)

The HxC firmware reads this image just fine but flashfloppy gives an error 31 on this image. The same image converted to an 80 track HFE image with double stepping will work on an 80 track machine (which will also do double stepping in the firmware) and this image is also accepted by flash floppy, but my 40 track machine obviously fails to read this image. For reference here is the accepted 80 track HFE file:
[1942.hfe.zip](https://github.com/keirf/flashfloppy/files/14780058/1942.hfe.zip)

keirf, 2024-03-27:
I just tried with FlashFloppy 3.42 on a SFRKC30.AT2 (AT32F415 chip) and both HFE images mounted fine for me. That's unsurprising since they differ in exactly one byte: The byte that tells the firmware to double step, or not.

sdsnatcher73, 2024-03-28:
Hmm. if I just put the 1 image on the USB drive it indeed seems to work just fine. If I put all of them on it giver error 31 and won't let me select any image... it shows 00/65535 Empty, if I press next image I get the error 31... should I flash the logging firmware to find out which image is causing issues?

sdsnatcher73, 2024-03-28:
Okay, turns out the directory I used as base had the HXCSDFE.CFG (which is needed on the HxC firmware) in it and that threw of the FlashFloppy firmware. Might be wise to ignore this file if present? Anyway case closed, error between the chair and the keyboard ;)

keirf, 2024-03-28:
That file puts FlashFloppy in HxC compatibility slot mode, in which locations of image files on stick are encoded in the HXCSDFE.CFG file. Of course you don't want that, and anyway the physical file locations are incorrect on your new stick. Hence FlashFloppy finds only garbage and throws an error.

I have added an item to the Troubleshooting page of the wiki.

closed 2024-03-28 by keirf