keirf/flashfloppy #887: ArteryISPProgrammer.exe problem

shadowc, 2024-03-31:
Hi! 

When trying to follow the steps for Artery device flashing, the program used to flash the device exits with the following error, which is not documented in the docs. Has anybody found this issue or know how to fix is?

[image]

I'd appreciate any help with this as I'm stuck!

keirf, 2024-03-31:
I've definitely had the program consistently crash on me before, but not recently. Usually I could update to a more recent version and that fixed it.

shadowc, 2024-03-31:
Thanks for the quick reply! Any chance to get an older version of it? I think the website only lists the last one!

keirf, 2024-04-01:
There's an older version on the HxC website https://hxc2001.com/docs/gotek-floppy-emulator-hxc-firmware/pages/flash-a-gotek-with-a-usb-cable.html

shadowc, 2024-04-01:
Thanks so much! I just tried with the older version (from hxc2001) and it worked! (I still need to try if the flash actually occurred and the device now works as an IBM floppy drive, but the program seemed to work perfectly).

Turns out there is a local `./Map` folder in the installation dir, but somehow the newer version of the program is looking for it at `C:\WINDOWS\system32`. My guess is extremely terrible programming and some `cd` command that changed the current directory wasn't cleared.

I do have a bit of a concern since I do not know any of these sites, but I chose to trust and scan the files with antivirus software. So if anyone is also wondering, I only have the OK from the antivirus as proof that stuff is OK to run. :)

Let me come back to you later with confirmation that the device is working so we can close this issue. I think this will be useful information for other people having the same issue!

keirf, 2024-04-01:
Yes the programming software is dodgy as, sadly. I do still use it myself from time to time, but I'm intending to set myself up with an ST-/J-/AT-Link programmer and OpenOCD on Linux instead if at all possible. At least their OpenOCD changes are published on GitHub.

shadowc, 2024-04-01:
Ok, I tried this on an IBM PC 386 with DOS installed and the firmware seems to be working (it created an IMAGE_A.CFG file for example) but I am unable to use it... When I switch to the drive (currently configured as drive B) I get a"insert a disk and press any key when ready", but I then see a "Not ready reading drive B".

I'm using FF.CFG to specify ibmpc interface and PC-DOS host but to no avail.

The weirdest thing is, when I press the key, it appears that the system is trying to read from drive A, which is an old 5"25 inch drive. I guess we can close the issue as the problem with the programmer software was solved, but it's a bummer I cannot make it read my images... :/ It's probably due to some issue with my BIOS/hardware configuration that I need to asses.

keirf, 2024-04-01:
Jumper at S1? You may as well jumper JC too. Does the Gotek green led light up when accessed? If not there's some basic drive select issue.

shadowc, 2024-04-01:
OK! With Jumper at JC and S1 I do get a "General Failure Reading Drive B" error... I suppose drivers may be swapped either in the cable or BIOS configuration. I will keep trying!

shadowc, 2024-04-01:
Update: When I take off the USB stick, I get a proper "No tready reading drive B", I created the test empty floppy image as:

```cmd
fsutil file createnew backup001.vfd 1474560
```

is it possible this command created a corrupted image? or it's not formatted correctly?

shadowc, 2024-04-01:
Yes, correct! My empty floppy disk image was no good!

I tried with a DOS 6.22 image and it worked like a charm!

Thanks so much for your constant help!! I'll close this! <3

closed 2024-04-01 by shadowc

keirf, 2024-04-01:
Great! One final thing to consider is returning to `host=unspecified` as for common IMG sizes (720k etc) FlashFloppy can determine the likely geometry itself and doesn't need to look at the image boot sector. But if everythign is working, maybe don't mess!