keirf/flashfloppy #890: Toshiba 110CS Pentium laptop not reading Gotek

zparras, 2024-04-14:
I just bought a Gotek with Oled and rotary selector (Flash Floppy 3.42).  I’m trying to get it working with my Toshiba Satellite 110CS Pentium 100 mhz laptop (from 1996). I am connecting the Gotek through a flat 26 pin cable (1.0mm pitch) connected to a 34 pin adapter. I’m running WIN 95C on this computer. I have a 2gb SD card in place of the old HDD and everything works fine except for the Gotek. The Gotek is powered and I can rotate thru the disk images. The light on the USB drive flashes when I try to access drive A: and the Track number flickers when I try to access it but I get drive not ready error.  I think my issue is with the specific settings with the FF.CFG file. I tried different variations of interface = shugart / ibmpc / ppc, host = pc-dos / pc98, pin02 = auto / chg, pin34 = rdy / chg

Does anyone have experience connecting a Goetk with a similar Toshiba laptop?

windship, 2024-04-16:
I think it's better to check jumper setting first.

zparras, 2024-04-16:
I have JC and S1 closed. I tested the Gotek on my Pentium XP desktop and it worked fine with the standard IBM PC settings. I suspect for my Toshiba laptop I need a different 26 pin to 34 pin  connector. I am using this adapter I bought from Ebay.
[image]

I am also looking to try these other adapters:
https://floppyemulator.com/products/34-26p1adaptor/

https://www.ebay.com/itm/304806696386?_trkparms=amclksrc%3DITM%26aid%3D777008%26algo%3DPERSONAL.TOPIC%26ao%3D1%26asc%3D20230811123856%26meid%3D10db4427773a45c38522790cf8f74f89%26pid%3D101770%26rk%3D1%26rkt%3D1%26itm%3D304806696386%26pmt%3D0%26noa%3D1%26pg%3D4375194%26algv%3DRecentlyViewedItemsV2%26brand%3DMarkenlos&_trksid=p4375194.c101770.m146925&_trkparms=parentrq%3Ae66a7d9e18e0ab3a170d8d24ffffca04%7Cpageci%3Ab3725235-fbda-11ee-91cc-da103003c0b3%7Ciid%3A1%7Cvlpname%3Avlp_homepage

Does anyone have any experience with either of these?

windship, 2024-04-17:
Did you try to connect a normal FDD using that adapter? Is it working?

windship, 2024-04-18:
Then it seems there's no problem with GOTEK. You better check adapters or FDD connector port first.

zparras, 2024-04-24:
Hi there, I have checked the traces from my adapter and I don't think I have the correct adapter. I've ordered a different one and will report back if it works.

zparras, 2024-05-31:
After more research I can report back that the Gotek works perfectly with my Toshiba 110CS laptop. I will summarize that I did so others with a similar computer (Toshiba laptop) can benefit from my experience.

After trying with the first adapter (pictured above), I ordered a second more expensive adapter (picture below) that still didn't work. I reviewed the Toshiba fdd controller pin out and realized the Ready signal was not connected through the adapter to the floppy drive. I've subsequently learned that modern floppy drives do not use the Ready signal and is not needed. I tried many different setting configuration on the Gotek CFG file with no success. I also learned from other posts that the Toshiba laptops (from this period) expect a Ready signal low (ground) so I modded pin 8 on the second adapter, cut the existing trace and grounded the Ready line. This did it! The Gotek worked with all the default settings for IBM PC (Settings below). Also with this mod the physical Panasonic Fdd work perfectly. I now have the choice now to use actual floppies or emulated. Also note I am not using the twisted cable to make the Gotek or physical FDD work.

For now the Gotek will hang outside the laptop on a temporary basis and be attached when I need it. Due to the limited space inside the floppy I might look for a second Gotek without the extra mods, remove the case and add the Gotek board inside the laptop. That will be for a future project, since I spent weeks on this project. For now I want to enjoy using a working floppy again.

In my research I learned that many Toshiba laptops from the late 1980s until the late 1990s used the Ready signal for it's floppy drives to work. I suspect there are many people with Toshiba laptops without a working floppy drive that could use some help.

Cheers!

List of the hardware I used:
Ribbon cable 26 pin 1.0mm pitch (from Amazon)
26 pin 1.0mm pitch to 34 pin adapter
Gotek / Panasonic 1.4mb fdd (one I had laying around)
Gotek settings:
Flash Floppy v3.42
Jumper JC01 & D01
interface = ibmpc
host = pc-dos
pin02 = auto
pin34 = auto

Toshiba FDD controller pin out 26 pin to 34 pin FDD
26 Pin to 34 pin

index 2 to 8 index
drive select 4 to 12 drive select
disk change 6 to 34 disk change/ready
ready 8 to ground (this is the only mod I did)
mode select 9 to 2 REDWC density select
motor on 10 to 16 motor on
direction select 12 to 18 direction select
step 14 to 20 step
write data 16 to 22 write data
write gate 18 to 24 write gate
track 0 20 to 26 track 0
write protect 22 to 28 write protect
read data 24 to 30 read data
side select 26 to 32 side select

Ribbon cable 26 pin 1.0mm pitch
[image]

26 pin 1.0mm pitch to 34 pin adapter
[image]

Pin 8 mod on adapter
[image]

Gotek working with Toshiba 110CS
[image]

windship, 2024-05-31:
Cool, congratulations for your success! 
So there's no problem with GOTEK, right? Users need to check the connection and jumpers first :)

I think you can close this issue now.

zparras, 2024-05-31:
Thank you for the support, I appreciate it. I closed the issue.

With kind regards

Zs

closed 2024-05-31 by zparras