keirf/flashfloppy #894: Akai S2000 .IMG files not working in my FF Gotek with supposedly correct settings

ipogeorge, 2024-04-29:
I'm not able to get my Gotek working with Akai .IMG files in my S2000.
The suggested settings (that I used and seem to be confirmed) should be these:
Akai:   The correct jumper selection is usually S0, with the following interface selections in FF.CFG, depending on model.
S2000, S3000XL: 
interface = ibmpc-hdout
host = akai

I tried lots of legit S900 and S2000-S3000 .IMG files but no one worked. 
I had an .HFE S2000 image and it was perfectly working.
So I converted all my .IMG files in .HFE files (using the available S3000 option) and they are working. 
Problem solved, more or less. But I still would like to understand why the .IMG files are not working.

Then, I tried lots of possible combinations (jumpers, host, interface): S0, S1, JC, ibmpc-hdout, S950, ibmpc, etc.. but no luck with .IMG files, still only .HFE ones are working with a few settings.

keirf, 2024-04-30:
The only setting that will make a difference for IMG vs HFE handling is `host = akai`. Can you zip up an offending IMG file, the corresponding working HFE file, and your FF.CFG file, and attach them here please.

ipogeorge, 2024-04-30:
[Akai S2000 files Github.zip](https://github.com/keirf/flashfloppy/files/15167678/Akai.S2000.files.Github.zip)
Many thaks for answering. Here are the requested attachments. 
Other useful details: 
- FF sees the .IMG files, but the S2000 says "Unknown format! or unformatted? ;
- The .HFE version is built from the same .IMG file;
- I've tried unsuccessfully other suggested settings found over the net, like pin02 = dens or autoselect-file-secs 3.

ipogeorge, 2024-04-30:
Another detail: the MS-DOS .IMG files are working with the sampler
So the issue is only with the S900-S2000 formats .IMG files.

closed 2024-04-30 by keirf