keirf/flashfloppy #897: Amiga Gotek

dekinikolic, 2024-05-14:
I have Amiga 500 with internal Gotek drive. Drive works, BUT if I load E.g. slots 19,20,21....with some multi disk game I hit F10 to save and restart, Amiga restarts but start booting slot 1 instead slot 19. So, my issue is that I cant force Gotek to boot from any slot other than slot 1.
Because of this there is no sense to load and save all available games in slots. 
On flash disk I have 2 files:
Autoboot.cfg
Hxcsdfe.cfg 
Any help??

keirf, 2024-05-14:
So if you set up all your slots, and don't use the file selector (slot 0) program again, you are saying you can't later make it boot from anything than slot 1? That doesn't make sense.

dekinikolic, 2024-05-14:
Thats right. I have loaded  20 slots with few games and apps I have, and if  want to play E.g. Toki (slot8), I select that slot (using keyboard, i am not touching those two buttons on Gotek),  hit F10, and Amiga restars whits Amiga test kit (slot 1).
I saw Chris Edwards is doing  exacly same  thing I want to achieve on his videol. 
Here is video [https://www.youtube.com/watch?v=aquNTxvAVSQ](url)
time stamp (13:29)
I am sendig screen of my Amiga setup-maybe it will help. I dont know......probably its my mistake. 

[image]

keirf, 2024-05-14:
Oh I see. I think the key you are looking for is F7. If you hit that in the "Drive A slots selection" screen then you should reboot into the currently highlighted slot.

dekinikolic, 2024-05-14:
Hm.....I dont think so.  If I hit F7 that option move adf to slot 1 and starts game....and remove adf which was previosly in slot 1. And if game is multi disk, than first disk after i hit F7 will be in slot 1 and others will stay on their previous slots. 
Am I wrong?

dekinikolic, 2024-05-14:
Main reason why I am trying to do this is that if some game is in slot E.g.150... I need to pres buttons on GOTEK 150 times to reach  that slot :))
Is there some other way to quicklu "jump" from E.g. slot1 to slot 205?

keirf, 2024-05-14:
With just buttons it is tough to accurately move such a large slot distance.

I am reading the file selector code rather than actually testing this, but: There are two main screens to the file selector, right? (1) The list of files on the USB stick; and (2) The list of assigned slots. By my reading of the code, F7 does just as you say on screen (1). But if you highlight a slot on screen (2) then it does look like F7 really should do the right thing.

dekinikolic, 2024-05-15:
FINALY!!!! it works.
As you said, solution is to highlight slot (with game in it) on screen 2 (list of assigned slots) and hit F7. Amiga will reset and game in selected slot will start. Now everything has much more sense. I can assign all games on my flesh disk to available slots, save everything (hitF9), and than from that list I can chose which game I will play.  
Thanks for your time....I hope that this will help others with same problem.

closed 2024-05-15 by keirf