keirf/flashfloppy #898: Camputers lynx

Gary-Clark, 2024-05-15:
Can native compatibility be included into FF for *.LDF files ?

Gary-Clark, 2024-05-16:
Working CFG file I am using

IMG.CFG

cyls = 40
heads = 1
secs = 10
bps = 512
id = 1
mode = mfm
interleave = 6
cskew = 0
hskew = 0
rpm = 300
gap3 = 0
iam = yes
rate = 250
file-layout = sequential

and

FF.CFG

interface = shugart
host = unspecified
pin02 = nc
pin34 = RDY

Works for the standard 200k drives (size 204,800 bytes),
FF should also be able to expect 80T double side images with a size of 819,200 bytes

keirf, 2024-08-23:
I note that this is from discussion thread https://forum.vcfed.org/index.php?threads/camputers-lynx.1247846/

Are all the images extension LDF, or is there also extension LDF8?

Did you get this working on Artery Goteks (I noted an IMG.CFG error in the other ticket you had open)?

keirf, 2024-09-03:
Any further info needed? Else I will close the issue shortly.

Gary-Clark, 2024-09-23:
Apologies, distracted by other machines at the NWCM :)

LDF files are the disk image files for the Lynx but its probably the error in the CFG file that caused the issue. Thanks for 

It would still be good if FF recognised a *.LDF file as a native image rather than having to include a CFG file