keirf/flashfloppy #902: SFR1M44-DU26 Pinout

victor7376, 2024-06-03:
Hi,

I've just been working on an old Toshiba T1850 laptop & with the help of a friend got a 34 pin GoTek to work perfectly by connecting a 26 pin to the 34 pin.

26 pin to 34pin (Toshiba to Gotek)
2 to 8 with 1k resistor 5v = Index
4 to 12 = Drive Select
6 to 34 with 1k resistor 5v = Disk Change
8 connected to ground = Drive Ready (toshiba needs it)
9 to 2 with 1k resistor 5v = Density Select
10 to 16 = Motor on
12 to 18 = Direction Select
14 to 20 = Step
16 to 22 = Write Data
18 to 24 = Write Gate
20 to 26 with 1k resistor 5v = Track 0
22 to 28 with 1k resistor 5v = Write Protect
24 to 30 with 1k resistor 5v = Read Data
26 to 32 = Side Select

I've just got hold of a SFR1M44-DU26 with board revision SFRC2D.B & would love to place this inside the laptop but I cant seem to find the pinout of the emulator for the 26 pin.

Could anyone give me some advice please.

Cheers

keirf, 2024-08-23:
The above is standard 26-pin micro-floppy pin assignment. It will match the SFR1M44-DU26 precisely. However you will need to modify this Gotek model to run it with FlashFloppy. See #155

closed 2024-08-23 by keirf

martinfarnan, 2026-08-31:
I have this almost working with my Toshiba 1850. The only issue remaining is that the dskchg is not working (dskchg is working on the 26-34 pin version i have been testing with....). On the 26-34 pin connector i have pin 6 connected to 34 and the FF.cfg file set to output dskchg on pin 34.(6 so it works). But on the 26pin Gotec, pin 6 goes to DSKCHG (PB6 on the controller). I think, it should be a FF.cfg change... but which settings.....I just figured it out when writing this...it needs to be set to shugart (P2 corresponds to Pin6 on the 26 pin variant of the gotek). It is working perfectly now...!   Keirf, this is brilliant work...now to get the harddisk working on the T1850....

victor7376, 2026-09-01:
> I have this almost working with my Toshiba 1850. The only issue remaining is that the dskchg is not working (dskchg is working on the 26-34 pin version i have been testing with....). On the 26-34 [...]

I have a sd card as my hard drive now, just had to buy a sd card adapter & flash the bios of the adapter for the T1850 to recognise it as a hard drive.
 If you’re interested in this option, the firmware is on vogons forum under ‘hdd replacement options’ for the T1850C