keirf/flashfloppy #903: Barbudan bent embroidery machine with ye-data YD-702D-6639D weird pinout

wiensmit, 2024-06-08:
Hi there, thanks for the great work.

Im trying to replace a ye-data YD-702D-6639D drive for 3.5 inch HSDS floppy's with a SFRKC30.AT2 controller

after some fooling around i think i found the reason why i cant seem to read files from the usb 
found this table [online](https://hansotten.file-hunter.com/do-it-yourself/msx-floppy-drives/fdd-repair/) and the drive seems to have a special pinout:
[image]

this also results into some funky number showing in the 3 digit display, see [video ](https://photos.app.goo.gl/aicejXnTTNDsEPfR6)

not sure how to configure this. tried the FF.CFG, but it doesn't have enough options for the pins.

keirf, 2024-06-09:
Also the jumper arrangement on the drive matters. See here http://jope.fi/drives/Y-E_DATA_702D_Series.pdf

How are yours set?

wiensmit, 2024-06-09:
mine is [chock full](https://photos.google.com/share/AF1QipNE5LdUiUsO_uhpR39dst3etw_fKivSk031gBGSzIrP8RwPkKt0qRM35h9VsIyfdg/photo/AF1QipMgnGWBmusN-Ye3aH3hbiv9UgM9mXUUYDxZ13KZ?key=YTRlZDZHMGJQYjFoM2xkdWhMWkVnWlRpOTU0c2VR) with jumpers:

according to the document you linked thats
H4
DC2
RY
DS0

keirf, 2024-06-10:
Okay using the "decoder ring" this means: Disk change on pin 2, and Ready on pin 34. This is `interface=shugart` in FF.CFG. Which is also the default. 

You will need a jumper on S0 because this drive is jumpered as drive 0.

Unfortunately it is also jumpered to output density mode on pin 4. Gotek can't do that automatically as it has no microcontroller connection to pin 4. However:
(a) you might be lucky and the host doesn't actually read pin 4
(b) the default high level of pin 4 will match either double- or high-density, and that might match the disk image you are using, anyway
(c) you can manually switch pin 4 to low level by adding a jumper at position J5 at the rear of the Gotek.

If you have messed a lot with FF.CFG I would also suggest deleting it, and resetting any cached config in the Gotek by holding both buttons for a few seconds while no USB drive is inserted, until the display changes.

wiensmit, 2024-06-10:
your awesome help did the trick. The jumper at J5 to be more specific.

closed 2024-06-11 by keirf