keirf/flashfloppy #904: Amiga External DF1 Gotek AT4.35 with FF v3.42 - FDD Disk change not working

mds-au, 2024-06-14:
Hi all - I have a new Gotek drive and also one that previously worked fine that now doesn't recognise a disk change request any longer. I have re-flashed the firmware, tried alternative versions of flash floppy, formatted the USB stick a few times (FAT32), tried different USB sticks, all to no avail.  

The Gotek will read and write fine (for the adf disk that is selected on boot) but it won't change when a new file is selected (via the rotary selector or via the buttons). 

I have tried the devices on 3 different Amigas (Amiga 500 WB1.3, Amiga 500 WB 3.2, Amiga 1200 WB 3.2.1).
I have a third Gotek that works fine on all three of the mentioned Amigas. 
Also tried different external port adapters for the Amiga (all of which work fine on the Gotek that I have that does work ok).

In addition to the standard required autoboot.hfe and hxcsdfe.cfg files on the root of the usb I have also tried forcing configuration settings via an FF.CFG file using: 
interface = amiga 
motor-delay = 500

But this also makes no difference. 

The only jumper I have set is on the S0 pins of the Gotek. 
Any suggestions as to possible causes and ideally a resolution would be very much appreciated.

Kind regards   Mark

keirf, 2024-06-14:
Disk change is pin 2, so you could check that pin voltage level on disk changes with a meter or scope. You can also boot Amiga Test Kit and then run the Floppy Signal test. When you eject a disk you should see `CHG` flashes up as an asserted signal, and that should remain so until you insert a new disk, and "step" the drive head.

Amiga should need little tweaking of Gotek out of the box. Actually I suggest *removing* FF.CFG entirely for now, and resetting any cached config by pressing the rotary for a few seconds, while no USB stick is inserted, until the display changes.

mds-au, 2024-06-14:
Thanks for the reply Keir. 

I don't have a scope but can ask around the local community to see if someone can assist. I have a basic multimeter so can try that. 

I have removed the FF.CFG and reset the Gotek a few times as well. I get the upside screen display but know how to resolve that with another config line if I recopy an FF.CFG file. No change with the issue continuing unfortunately.

I have booted with the latest Amiga Test Kit disk but don't see any CHG wording come up. 

Should I be seeing this on the Test Kit software or on the OLED screen of the Gotek ?

On the Amiga in the F3: Floppy Drive section I can DF0 and DF1 are seeing DS-DD 80 and no change if I select DF1 and try changing files on the Gotek.

F5 signal test and selected DF1 says Drive READY too fast (208us): Gotek or modified PC Drive ?
It completes the read with all green 0's 

No change to the READY when selecting a new file on the Gotek.

On the Gotek OLED Screen - when selecting the new file it comes up with ADF ahead of displaying the track details.

keirf, 2024-06-14:
CHG will appear on ATK screen alongside other signals such as TRK0 (when on track zero), RDY (when image is mounted), etc.

It sounds like your pin 2 outputs have got damaged, somehow. You may need to swap out the relevant buffer transistor for a '3904 NPN transistor. Very cheap. Easy. If you can solder SMT.

keirf, 2024-06-14:
What model are these Goteks by the way (printed on PCB at rear)

mds-au, 2024-06-14:
SFRKC30.AT4.35

keirf, 2024-06-14:
Oh the nice 435-based ones. How sad they have seemingly gone wrong :( I wonder what might have caused it. Perhaps static, depending on whether the output transistors are FETs vs BJTs (either can work).

keirf, 2024-06-14:
Probably worth finding someone local who can help you test pin 2, and do the fix (very easy for someone with the skills). If you were UK based I would offer.

mds-au, 2024-06-14:
re ATK I can understand what you meant now regarding what should be seen when running the Floppy signal test as I did this with the Gotek that is working and I could see it cycle through as you mentioned with CHG, TRK0 and RDY etc.  So yes it does unfortunately look like these drives have a hardware issue. 

Thank you for your assistance and kind offer to assist and I was able to get the units to you I would but being based in Australia the logistics and costs are a likely going to make it not so cost efficient. 

I will try seeking someone local as you have suggested and put some posts to local usergroups etc. and see if I get any responses.  Thanks again.  Best wishes   Mark

keirf, 2024-06-14:
One very final suggestion: Can you check the 5v line with your multimeter? Goteks don't like much more than 5v. Anything up to about 5.5v should be fine.

closed 2024-06-17 by keirf