keirf/flashfloppy #906: MT-100 Pinout different than in documentation

Pandorify, 2024-06-23:
Hi, it seems the instructions for the QuickDisk wiring on the Roland MT-100 is wildly wrong on the wiki, in a way that could send 5v to data pins and damage the Gotek. I was able to find what I believe to be a proper pinout mapping, which was confirmed as working by other users and seems to be working for me as well, on the HxC forums here: https://torlus.com/floppy/forum/viewtopic.php?p=21542#p21542

[image]

keirf, 2024-06-23:
Thanks I will put this in the wiki.

keirf, 2024-08-23:
Done. Please reopen if you see any issues in the revised documentation.

closed 2024-08-23 by keirf