keirf/flashfloppy #908: OS/2 Warp 3 files give ERR 31

ltning, 2024-06-25:
Hi,
been trying to get FF (both latest 3.x and 4.x branch) to read the floppy images from OS/2 Warp 3.0, with no luck.
Some files are 1.44MB, others 1.8MB. The 1.8MB have a header reading
```
XDF v1.1a (c) 1993, 1994 -- Backup Technologies, Inc., Tampa FL
Patent(s) Pending -- Inventor: Roger D. Ivey
```

The 1.44MB images don't have this header, but the BSD `file(1)` command returns (showing both file types here)
```
disk1.img:    floppy image data (IBM SaveDskF), 2848 clusters, checksum 0x2eafa6fc, sectors 2880, 1st sector at 0x29
disk10.img:   floppy image data (IBM SaveDskF), 1st cluster at sector 37, 3644 clusters, sectors/FAT 11, root directory at sector 23, checksum 0xa272a927, sectors/track 23, sectors 3680, 1st sector at 0x29
```

I've tried in a 32K and 384K Gotek model. Other images work fine. I've tried .xdf, .img, .dsk suffixes. I'm at a loss, sadly.

The two files are attached here.

I'd try to write them to HFE images but I can't for the life of me figure out how to create one - I don't have Windows nor Macs available to me, and haven't found any UNIX-y utilities (I'm sure they exist, I'm just blind).. Halp?

[ff-os2warp3.zip](https://github.com/user-attachments/files/15978288/ff-os2warp3.zip)

keirf, 2024-06-26:
These are indeed `SaveDskF` image files, and not directly supported by FlashFloppy. Actually there's very little modern software that does support them. I found https://github.com/aaru-dps/Aaru which may allow conversion to IMG or XDF. And then we would be in business. But these supplied images as-is -- no chance, mate!

keirf, 2024-06-26:
By the way, if you get stuck then consider raising a ticket on the Aaru project. The author @claunia is very helpful and helped FlashFloppy project previously. They know a lot about XDF!

ltning, 2024-06-26:
Thanks, I'll check the Aaru project.
I found that if I chop off the first 41 bytes of the images, FF is willing to serve them up. For the 1.44MB images, that seems to work fine, but for the larger ones the OS/2 installer doesn't recognize them despite FF not giving an error.

It should also be said that the typical tools used to write these images to floppies - LOADDSKF and XDFCOPY - both refuse to work with the files I sent you. However after stripping the first 41 bytes off, LOADDSKF will write the smaller one - and the resulting floppy/FF image is readable. For the larger file, XDFCOPY will write the file to floppy (only tested using your high_density.hfe) - but fail to verify.

I realised the 41 byte issue when comparing these images with images found on a different OS/2 CD. I suppose I should also check if there's anything at the end of the image that needs trimming, but it doesn't seem that way..

keirf, 2024-06-26:
I think the format is a bit fancier than a mere 41 byte header, at least where XDF images are concerned. So you've got a bit more legwork to do there! It's definitely a promising start though.

keirf, 2024-08-23:
Closing as a "won't fix" for FlashFloppy.

closed 2024-08-23 by keirf