keirf/flashfloppy #909: Errors with new F415 & F435 Goteks

Gary-Clark, 2024-06-27:
Having problems with flashfloppy on the new Goteks

Working with a Camputers Lynx interface which works fine with a SFRC922D Gotek but fails with a SFRKC30.AT2 or the .AT4.35 Goteks (with the latest stable version). I have tried different USB drives and even a different interface / Computer.

Some images give a directory, others just return a 'DOS &C3' error

Ran a format on an image with the log version and got the log below. I note the [lost 4597] looks suspicious but not sure how to read the log.

Build: Jul 29 2022 14:19:26
Board: Standard
Config: Flash Slot 10 (ver 2, size 70)
Interface: Shugart (pin2=high, pin34=rdy)
I2C: SDA held by slave? Fixing... Done
I2C: OLED found at 0x3c
OLED: SSD1306
Display: LCD/OLED

> Device speed: Full
 VID : 14CD
 PID : 1212
> Class connected: 08 (MSC)
 Manufacturer : Generic
 Product : Mass Storage Device
 Serial Number : 121220160204
Cache 718 items
Config: Erased Slot 10
Config: Written to Flash Slot 11
Mode: Native
Cache 718 items
Current slot: 0/1
Name: '96K20cln.LDF' Type: img
Attr: 20 Clus: 00000008 Size: 204800
Fast Seek: 1 frags
Cache 718 items
Before: After: Current slot: 1/1
Name: 'Games 1.LDF' Type: img
Attr: 20 Clus: 0000000f Size: 204800

[lost 4597]
rite 3[04]/10
Write 8[09]/10
Write 4[05]/10
Write 9[0a]/10
Write 0[01]/10
Write 5[06]/10
Write 1[02]/10
Write 6[07]/10
Write 2[03]/10
Write 7[08]/10

Gary-Clark, 2024-06-28:
Converting the LDF files to HFE and now they load on the new Goteks ?

Odd

keirf, 2024-08-20:
What's an LDF file? Or are these images named something like `name.LDF.IMG`?

keirf, 2024-08-20:
Oh I found your thread here https://forum.vcfed.org/index.php?threads/camputers-lynx.1247846/

Your IMG.CFG is wrong at least as stated in that thread. The correct form of heading would be `'[LDF::819200]` rather than `[*.LDF::819200]`. If the latter ever worked, it is surprising.

keirf, 2024-08-23:
Please reopen if issues persist.

closed 2024-08-23 by keirf

Gary-Clark, 2024-09-29:
Thank you.

Clears things up nicely