keirf/flashfloppy #910: GOTEK and MSDOS - 1.44MB disks do not work

arti040, 2024-07-03:
I got config as follows:
S1 jumper - closed,
JC jumper - closed,
J5 jumper - closed (for IBM PS1 2011 model and high destiny drives... but I does not change anything anyway)

FF.cfg:
interface = ibm-pc
host = "unspecified" or "pc-dos"

Problem is, my PC recognizes it only as 5.25/1.2MB and boots only from such images.
Tested on two GOTEKs and few USB drives + did some tests on more standard, 486-class computer - the same effect. Only 5.25 drive works.

keirf, 2024-07-03:
It's probably a BIOS option to select the attached drive type?

arti040, 2024-07-03:
On 486 yes - but only 5.25 variant works properly. On IBM PS1 I have no option to change anything. It just discovers such drive - and that's it. However, if I connect GOTEK to the second port on a cable ribbon, it discovers two drives - 5.25/1.2MB and 5.25/360KB... or two 5.25/360KB - depends on cable. I'm not able to force any of my computers to recognize it as 3.5/1.44MB drive. :/

keirf, 2024-07-03:
There is no more configuration nor identification, however. FF will automatically act as a 1.44M 3.5 drive when used with a 1.44M IMG file.

arti040, 2024-07-04:
Do you think that reflashing firmware would help?

keirf, 2024-07-04:
Are you definitely running FlashFloppy already (says 'F-F' or 'FlashFloppy' on display when no USB drive is inserted)?

arti040, 2024-07-04:
Yup. v. 3.42, I guess.

keirf, 2024-07-04:
I would concentrate on 486 to start with, as that will be a regular PC with no IBM funniness.

Remove your FF.CFG. You don't need it for regular PCs. Delete any cached config by pressing both buttons while no USB drive is inserted, for several seconds, until the display changes (will say 'RST' or 'Reset Flash' or something like that).

Keep jumpers at S1 and JC. Remove J5.

BIOS configure drive as 3.5/1.44 and try a bootable 1.44 image. It should work.

arti040, 2024-07-04:
Ok, will try and let you know later. Thanks!

arti040, 2024-07-04:
OK, I'm coming back with the status:
1. On 486 - it works perfectly fine as 3.5/1.44MB;
2. On IBM PS1 - when J5 is removed - it doesn't work - it tries to read first sector and returns and error - "Not ready drive A", however if I close J5 jumper, it does a few attempts by reading sectors 1.0, 2.0/2.1 and then stops with no errors.

keirf, 2024-07-04:
IBM PS/1 and PS/2 may use other pins to identify the floppy type. They aren't necessarily well supported.

arti040, 2024-07-04:
Yay! It works! I had to short 4th line of the ribbon cable to GND and that's it. Here are the details -> https://www.youtube.com/watch?v=XZb9WN_u0YA&t=700s

keirf, 2024-07-04:
That's great! I think shorting pin 4 is the same as jumpering J5 actually....

closed 2024-07-04 by keirf

arti040, 2024-07-04:
I got it shorted but if it's open, IBM discovers GOTEK as 3.5/720KB... strange.