keirf/flashfloppy #915: Korg Triton - Showing either No Disk or Unformatted

MikeyMcG, 2024-07-15:
I just got a gotek with artery at32f435 chip, and flashfloppy+ v3.42 preinstalled, to replace the floppy drive in my Triton Rack. It has an oled screen and jog dial. I have set the jumper to S1 as the Triton uses a standard ibmpc floppy drive. In the ff.cfg file, I set the interface to "ibmpc", the host type to "akai", and display-type to "oled-128x64-rotate", just to be on the safe side. 

After some trial and error creating floppy disk images, I was able to create them using poweriso, with an image size of 1.44mb and fat32 format. The gotek now "sees" the img files, and "opens them". But before "opening" an img, my triton reports there is no disk in the drive. And when I "open" the img, it reports the drive is unformatted.

Am I (obviously?) doing something wrong? Are there any further steps I need to take when creating the img files, to let the triton see the open img? Or am I going totally the wrong direction using flashfloppy?

keirf, 2024-07-15:
Floppy disks are FAT12 not FAT32. It would be correct that there is no disk in drive when no img is selected. Once selected, the IMG is "inserted" and the Triton will read it. It should be satisfied if it sees a valid FAT12 filesystem.

keirf, 2024-07-15:
I do still suggest diskflashback. It's integrated mostly into Windows Explorer so clicking on it won't bring up a GUI. If you look in the system tray in the taskbar you should see it in there, and from there you can create and mount new disk images. I'm looking at the "Creating Black Disk Images" section of the [tutorial video](https://www.youtube.com/watch?v=zFYf9jhp3bs)

MikeyMcG, 2024-07-16:
Hi, thanks for that. I watched the diskflashback video and then created the image files as high density, ibm pc, FAT12. I opened each image file on my pc and copied the files across. Again, the Gotek sees and opens the images, but the triton shows the same issues.

keirf, 2024-07-17:
Try a double density 720k pc fat12 IMG.

keirf, 2024-08-11:
Any luck?

MikeyMcG, 2024-08-11:
Unfortunately not.

-------- Original Message --------

keirf, 2024-08-11:
It will be something about the electrical connection to the Triton, or configuration of pin 2. You can try setting, in FF.CFG file (https://github.com/keirf/flashfloppy/wiki/FF.CFG-Configuration-File), `interface=ibmpc-hdout`.

Note that this is mentioned in the appropriate part of the wiki: https://github.com/keirf/flashfloppy/wiki/Host-Platforms#korg

MikeyMcG, 2024-08-11:
Thanks

I'll have a look once I can get a locksmith out. The lock on my studio door has seized up and no amount of lubrication or tapping gently with a hammer is convincing it otherwise.

-------- Original Message --------

keirf, 2024-08-23:
Closing tech support tickets. If you have further issues please use forums, Facebook group, or Discussions tab here on GitHub.