keirf/flashfloppy #916: GoTek Drive Incompatibility with USB Switch/Hub

Pontiac76, 2024-07-16:
Description:
I recently acquired a USB switch (not a hub) that allows me to connect four USB devices and toggle which of two PCs can access them. When I connect my USB key through the switch, my Windows 10 PC can detect it without any issues. However, my GoTek drive fails to recognize the USB key when it is connected via the switch.

Details:

    The USB switch enables switching between two PCs to see the connected USB devices.
    The four ports on the switch act like a USB hub.
    The GoTek drive does not seem to accept the USB key through the switch.

Request:
Please consider adding support for the GoTek drive to recognize USB keys when connected through a USB hub. This could include:

    Enabling the GoTek to read a single drive when connected through a hub.
    Adding functionality for the GoTek to detect multiple drives and allow selection of the desired USB key.

Setup:

    USB Switch with 4 ports.
    Windows 10 PC (recognizes USB key via switch without issues).
    GoTek drive (fails to recognize USB key via switch).

keirf, 2024-08-11:
This is probably unlikely. The USB stack being used doesn't support hubs, so it's a bunch of work and noone else has asked for this.

Pontiac76, 2024-08-16:
I managed to find a USB switch that doesn't act like a hub, but an actual pass through.  So this particular ticket isn't needed anymore.

closed 2024-08-16 by Pontiac76