keirf/flashfloppy #922: d2m d4m support for CMD FD-2000 / FD-4000

voutti, 2024-08-05:
Not sure if this is actually possible - but would be cool to have support for these containers.

markusC64, 2024-08-16:
It would be interesting to know if the following work around enabled you to use D2M images:

Rename the image file from *.d2m to *.cmdfd2000.img and use the following "img.cfg":

```
`[cmdfd2000::1658880]
cyls=81
heads=2
file-layout = sides-swapped,interleaved
   secs = 10
   bps = 1024
   id = 1
   mode = mfm
   interleave = 1
   gap2 = 22
   gap3 = 100
   iam = no
   rate = 500
tracks = 0-80.0
   h = 1
tracks = 0-80.1
   h = 0
````

I agree, support with the usual extension would be great. But on the other way I am pretty sure that it would help Keir to know if this configuration works (it should work).

keirf, 2024-08-16:
Are D2M and D4M image files raw 1.6m and 3.2m disks respectively?

markusC64, 2024-08-16:
Yes, they're. From the flux dump (in the greaseweazkle issue tracker you have sample images and a feature request) I conclude that this defition should work:

`
[cmdfd0200dd::829440]
cyls=81
heads=2
file-layout = sides-swapped,interleaved
   secs = 10
   bps = 512
   id = 1
   mode = mfm
   interleave = 1
   gap2 = 22
   gap3 = 51
   iam = no
   rate = 250
tracks = 0-80.0
   h = 1
tracks = 0-80.1
   h = 0

[cmdfd4000::3317760]
cyls=81
heads=2
file-layout = sides-swapped,interleaved
   secs = 20
   bps = 2048
   id = 1
   mode = mfm
   interleave = 1
   gap2 = 22
   gap3 = 100
   iam = no
   rate = 1000
tracks = 0-80.0
   h = 1
tracks = 0-80.1
   h = 0
`

markusC64, 2024-08-16:
You see, cmdfd4000 is d4m, and cmdfd2000dd is d1m.

voutti, 2024-08-18:
Thanks for these suggestions!
I downloaded a set of d4m images from:
[archive.org](https://archive.org/download/unrenamed-files-commodore-8bits/UnRenamed%20Files%20-%20Commodore%208Bits/Commodore%20C64%20files%20%5BD4M%5D.zip)

I tried with both "FD4000 Native.d4m" and "Metal Dust (2005)(Protovision).d4m" by changing the d4m extension to "cmdfd4000.img".
Powering ON the FD-4000 with Gotek (3 digit display, flashfloppy 3.42) and USB stick attached with the IMG.CFG as shown by markusC64 I see the display going through a sequence: " 0_", " 0-", " 0_" and then the Gotek stops into a state where it just displays " 0_".
Trying to load directory on C64 from FD-4000 gives immediate result "?FILE NOT FOUND ERROR". The green LED in GOTEK light for ~10 seconds.

markusC64, 2024-08-18:
Oh, well, my Flashfloppy is to slow for ED Density in this configuration (I conclude that from the symptoms). That is why I have asked you to try a D2M.

voutti, 2024-08-18:
Ok, I tried with a D2M. Still getting the same errors. However, the Gotek display does not get stuck into " 0_"  anymore. Instead it finally returns back to showing "000"

markusC64, 2024-08-18:
Well, I cannot explain that - the above configration works for me with a flashfloppy. I used a greaseweazle to check that the floppy emulation returns the correct MFM stream.

voutti, 2024-08-18:
I wasn't sure if I was supposed to test with the first or the second IMG_CFG -version - since the more recent one was a bit different for D2M. So, I checked also with the earlier one but that didn't make a difference.... I could be doing something wrong here, though....

markusC64, 2024-08-18:
The second one does not contain a definition for "d2m" alias cmdfd2000. It wasjust to tell Keir the definition of the disk layout and is untested.
cmdfd2000dd is what you get when you insert  DD disk into that drive. Resulting in a D1M image file.

So for HD disks in the FD 2000 and thus D2M you should use the 1st img.cfg. Or you can simply place both definitions into one single file.

voutti, 2024-08-18:
Since FD-4000 is compatible with 1581 disks as well I tried with a d81 - but still with the same errors....
Could this be something related to ED floppy drives having some different signalling for something? At least there is that detection for the ED hole which normal floppy drives don't have, for example....

keirf, 2024-08-18:
Are these images always/usually 81 cylinders? Actually I can support d2m/d4m directly in FlashFloppy I think, and get this auto-detection right, no problem. I can make D81/D2M/D4M all map to the d81 format table, and distinguish between them by file size. How does that sound?

markusC64, 2024-08-18:
Yes, they're always 81 cyliders. Since cylinder 80 (zero based numbering) contains the partition table, there is no way of having less. 
And since d?m cannot represent copy protections, we don't need to think if there is any that uses track 81 and above...

A d81 can have 81 tracks, too. Geos Wheels uses Track 80 (zero based) on a 1581 resp. a FDx000 in 1581 format for copy protection. But a d1m and a 81 track d81 are more or less the same, just targeted for a different floppy drive.

voutti, 2024-08-18:
> @keirf @voutti I think Timo's problems might have wth to do with your ff.cfg configuration - since I don't have another idea.

To be noted that currently I don't have ff.cfg on the USB stick.

keirf, 2024-08-18:
FF.CFG gets cached on Gotek. Reset cached config by pressing both buttons for several seconds with no usb stick inserted until the display changes.

voutti, 2024-08-18:
Ok, got a "rst" on display - so I assume that got rid of any previously cached FF.CFG...
This didn't change the behavior, though... So, perhaps I should have somehow modified FF.CFG.
... Just not sure if the "ED" things should be set there somehow. I had a look at the jumpers in my original FD-4000 floppy drive (TEAC FD-235J) and tried to decode the settings. I'll attach an image here with all the possible settings and the settings that have actually been selected are marked with a green ball.
[image]

voutti, 2024-08-18:
... I put the following pin 2 & 34 settings and got a directory read ok from d2m :)
pin02 = dens
pin34 = chg

keirf, 2024-08-18:
Great. Those pin settings are exactly equivalent to interface=ibmpc-hdout

voutti, 2024-08-18:
Yes, definitely nice that d2m seems to be working!
d4m still locks up the Gotek, though....
TEAC-235J seems to signal HD IN to pin6 and ED IN to pin 2. I don't really have a clue how this should be handled.... no option in FF.CFG for pin6...
And then there is that HD IN / ED IN level assignment settings (MSC with TEAC... what ever that means....).
Or is it as @markusC64 said that flashfloppy/Gotek lacks the required speed to handle ED traffic?

markusC64, 2024-08-18:
Fine. Now that D2M is working, there is no doubt that the firmware can be improved to work with the usual d2m extension.
No reason to neglect d1m, in the FDx000 you can insert DD disks... But since its dara rates are even lower, I would not expect problems at all.
And D81 shoudl also work out of the box.

markusC64, 2024-08-18:
I have been told that the FD4000 does not use the ED IN signal. You could even format a DD disk to ED with it. So it has to be something different.
With a gereaseweazle connected to the Gotek (well, I did that some years ago) I have seen that in ED mode I just got garbage with sector size 1k, With sector size 512 ED worked, but that is wthe wrong configuration for FD4000. Sector size 512 is corrcect for PC formatted ED disks.
Different Goteks exists, and since it's not an FD4000 isse, Keir might be able to explain that or look at that.

markusC64, 2024-08-18:
One thing that might make a difference: What Microcontroller does your Gotek have? Compare. https://github.com/keirf/flashfloppy/wiki/Gotek-Models

voutti, 2024-08-18:
The one I'm using has a STM32F105

markusC64, 2024-08-18:
I'd bet it's too weak to extra high density - and I hope Keir will tell us more.

BTW: My Gotek has the same microprocessor. But I am not sure that I took that Gotek when I did the ED test.

keirf, 2024-08-19:
I think ED read/write ought to work with STM32F105, but not formatting. Best is to use AT32F435 if possible.

voutti, 2024-08-19:
I ordered a new Gotek that should have AT32F435 on it... I needed another one, anyways.
I'll check with that when it arrives.

keirf, 2024-08-20:
Please can you try the following test firmware, which supports D2M and D4M image files directly:
https://github.com/keirf/flashfloppy/actions/runs/10474362108

You will need to be logged in to GitHub to see the Artifacts link at the bottom of the linked page.

voutti, 2024-08-20:
Holy smoke, it works !! 
...I mean both d2m and d4m work with this STM32F105. I put 1 of each file types on the USB stick and at least loading a prg from d4m worked just fine :)

markusC64, 2024-08-20:
See above, that is the format that cmdfd2000dd describes, it is identical to D81 from most points, but 81 Track and intended for CMD FDx000. Since it has a partition table on the 81th track that a 1581 cannot process.

markusC64, 2024-08-20:
Thanks. BTW: My ZIP files in https://github.com/keirf/greaseweazle/issues/471 contain sample images, not just the scp but also the corresponding d?m files.

keirf, 2024-08-20:
Thanks. I have also added support for these image types to Greaseweazle just now.

keirf, 2024-08-20:
Please reopen if there are any issues.

closed 2024-08-20 by keirf