keirf/flashfloppy #923: PC Not Recognizing Floppy

lechuck333, 2024-08-21:
I have a Windows 98 PC. I've flashed my Gotek with FlashFloppy, and it appears to have have worked. However, the PC doesn't recognize the drive.

The PC has a Pentium 2 CPU, Voodoo 3 GPU, Sound Blaster Live! sound card, a CF card adapter for a HDD. When I boot the PC, it shows the CF card as Master and the DVD drive as a slave, but it does not show any other devices (Gotek is set to slave). I even tried reversing the ribbon cable in the Gotek (the OLED screen said the cable was reversed, showing that there is a connection here). I set the interface to IBMPC in the config file. Not entirely sure where to go from here.

Any help would be appreciated :)

keirf, 2024-08-21:
All you should need is a jumper at S1 and a jumper at JC. Place the drive after the twist in the floppy drive cable to make it drive A. No config needed.

lechuck333, 2024-08-21:
> All you should need is a jumper at S1 and a jumper at JC. Place the drive after the twist in the floppy drive cable to make it drive A. No config needed.

I have those jumpers set. But still the PC doesn't detect the drive.

keirf, 2024-08-21:
Is floppy drive enabled in bios.

lechuck333, 2024-08-21:
Yes, it is enabled.

keirf, 2024-08-22:
Attach your FF.CFG here. You shouldn't even need one, actually.

S1 + JC should suffice, and then stick 1.44M/720M disk IMG files on your USB drive. Done.

Beyond that, not much help I can give. I would check your cable and also wonder about your floppy interface on the mobo.

keirf, 2024-08-22:
If you do delete your FF.CFG, then you need to reset any of it that is cached inside your Gotek. To do this, press both buttons while no USB drive is inserted, until the display changes (says 'RST' or 'Reset Flash Config'). A few seconds later your Gotek will reset and be factory fresh.

lechuck333, 2024-08-22:
## FF.CFG: Example FlashFloppy Configuration File

# Place in the root folder or FF/ subfolder of your USB drive.
# NOTE: If FF/ exists, IMG.CFG must reside there, not the root folder.

# Uncommented lines below are the default settings.
# Uncommented options cannot be overridden by settings in other config files.

##
## DRIVE EMULATION

# Floppy-drive interface mode (interface pins 2 and 34)
# jc: Specified by jumper JC (open: shugart, closed: ibmpc)
# shugart: P2=DSKCHG, P34=RDY (Amiga, Atari ST, many others)
# ibmpc: P2=unused, P34=DSKCHG (IBM PC interface)
# ibmpc-hdout: P2=HD_OUT, P34=DSKCHG (not generally needed: prefer 'ibmpc')
# jppc: P2=unused, P34=RDY (Japanese PC standard)
# jppc-hdout: P2=HD_OUT, P34=RDY (Japanese PC alternate: prefer 'jppc')
# akai-s950: Legacy alias of 'jppc-hdout', previously used for Akai S950
# amiga: P2=DSKCHG, P34=DRIVE_ID (not generally needed: prefer 'shugart')
interface = ibmpc

# Host platform: Improves image-format detection for generic types such as IMG
# acorn: Acorn ADFS
# akai: Akai (S01, S20, S950), Korg, SC Prophet 3000
# casio: Casio (FZ-1)
# dec: DEC (RX33, RX50)
# ensoniq: Ensoniq (ASR, TS, etc)
# fluke: Fluke 9100
# gem: General Music (S2, S3)
# ibm-3174: IBM 3174 Establishment Controller
# memotech: Memotech
# msx: MSX
# nascom: Nascom
# pc98: NEC PC-98
# pc-dos: PC DOS Format (geometry determined from Bios Parameter Block)
# tandy-coco: Tandy Color Computer (CoCo)
# ti99: TI-99/4A
# uknc: UKNC / DVK Soviet PDP-11
# unspecified: Common default geometries (including IBM PC)
host = unspecified

# Pins 2 & 34 output (drive->host) manual configuration
# auto: Auto-configure from interface= setting
# nc: Unused/No Connection [eg. if pin is an input (host->drive) on your host]
# low: Always 0 volts (0v)
# high: Always 5 volts (5v)
# rdy: Drive ready (Ready = 0v)
# nrdy: Logical complement of above
# dens: Density mode (High Density = 0v)
# ndens: Logical complement of above
# chg: Disk changed (Changed = 0v)
# nchg: Logical complement of above
# Values: auto, nc, low, high, rdy, nrdy, dens, ndens, chg, nchg
pin02 = auto
pin34 = auto

# Forcibly write-protect images, or respect the FAT read-only attribute?
# Values: yes | no
write-protect = no

# Maximum cylinder that can be stepped to (255 is required for access to
# Direct Access mode as used by image-selector utilities and Autoswap games).
# Values: 0 <= N <= 255
max-cyl = 255

# Filter glitches in the SIDE-select signal shorter than N microseconds
# Values: 0 <= N <= 255
side-select-glitch-filter = 0

[... 196 lines cut ...]
nav-scroll-pause = 300

##
## MISCELLANEOUS

# Speaker volume for head STEP
# Values: 0 <= N <= 20
step-volume = 10

# Speaker volume for insert, eject, and slot-number notifications.
# Slot number is indicated by a sequence of beeps when an image is mounted
# iff "slotnr" is specified. The slot number is then notified by a sequence of
# long beeps (each counting +5), followed by a sequence of short beeps
# (each counting +1).
# Values: N[,slotnr] (0 <= N <= 15)
notify-volume = 0

# Report the specified version number to host software
# Values: <quoted-string> ("" means report real version)
# eg. da-report-version = "v3.0.0.0"
da-report-version = ""

# Automatically extend certain types of truncated image file (SSD,DSD,TRD)?
# Values: yes | no
extend-image = yes

lechuck333, 2024-08-22:
Also did a factory reset. No difference unfortunately. Unless you see something in the config, I'll have to dig up a new floppy cable.

keirf, 2024-08-22:
I don't see anything wrong in the config.

keirf, 2024-08-23:
Closing as not a bug. You can get further tech support advice on forums, Facebook, or Discussions tab here in GitHub.