keirf/flashfloppy #924: PCW8256 Loading issues

A-little-confused, 2024-08-21:
Hello,
I am trying my hardest to get my Gotek to work on my PCW as drive A:
I have the jumpers on S0 and JA, an adapter to go from the PCW to the back of the Gotek, 5v and ground pins to the Gotek and can see when powered up on the Goteks OLED that DSK files are present. But still cannot get anything to load.

What am I doing wrong or not doing.

Many thanks

keirf, 2024-08-21:
Do you mean jumper on JA? What firmware is running on your Gotek?

A-little-confused, 2024-08-21:
Yes, Jumper JA and 4.7a firmware.

keirf, 2024-08-21:
Do not use the alpha firmware. Stick with 3.42.

Why do you have a jumper at JA? In fact this should stop the Gotek from initialising, as it will look like a button is permanently pressed (and put the Gotek in firmware update mode).

A-little-confused, 2024-08-21:
Right, I have tge jumpers on S0 and JC. Downgraded to 3.42 following the steps online and can see it briefly flash up when starting the machine that it is now at version 3.42.

Have cycled through the DSK files on the USB and clicked to have it launch, but still no loading, just the beeps (they indicate no disk) as we had before.

Sorry all

A-little-confused, 2024-08-21:
All this help is appreciated.
Have tried all of the jumper settings. But to no avail.

[image]
[image]

Maybe I don't have the correct CFG or HFE files. 

Thanks

keirf, 2024-08-21:
Is the adapter the right way round? I would expect pin 1 to pin1, and ribbon stripes matched to pin 1 at both ends as well.

keirf, 2024-08-21:
Okay, and that is definitely a PCW adapter? There are multiple 26-pin floppy interface layouts. Amstrad themselves used more than one afaik.

A-little-confused, 2024-08-21:
Right problem solved.

Set jumper to MO
Removed JC
Moved the ribbon and adapter to match 1 on all.

AND BOOM ITS WORKING

Thank you over and over again.
[image]
[image]

keirf, 2024-08-21:
Trial and error wins again!

closed 2024-08-21 by keirf