keirf/flashfloppy #932: 2DD writing problem with external FDD of PC-9801

RGM-179, 2024-09-13:
Nice to meet you.

I find Flashfloppy very useful.

I attached an external FDD and 2DD control board to my PC-9801RA.

It started up without any problems, but when I play a 2DD game and save it, it seems to finish normally, but the disk gets corrupted.

I'm using it with 

FF.CFG

interface = jppc
or
interface = jppc-hdout

host = pc98

pin02 = auto
pin34 = auto
.

I've tested the games SORCERIAN and HYDLIDE.

With SORCERIAN, the user disk itself gets corrupted and becomes inaccessible.

Do you have any ideas?

keirf, 2024-09-13:
What's the image type of your 2DD disks (HFE, IMG, etc)? What's your Gotek type (MCU and/or PCB name)? USB drive model name? If it's an AT32F415 based Gotek, try getting an AT32F435 based Gotek or try another USB drive.

RGM-179, 2024-09-13:
Thank you for your reply.

The Gotek I am using is as follows:

SFRKC30.AT3

AT32F415

The file format is "hfe".

I will try looking for the AT32F435 type.

keirf, 2024-09-13:
Or see if you can switch to an IMG type. It needs less buffering for writes.

RGM-179, 2024-09-13:
I tried converting it to img, and it seems to load, but never finishes.

I have D88 and NDF images, but even when I converted them to FDI, they didn't load properly.

RGM-179, 2024-09-14:
Thank you for taking time out of your busy schedule.

Just to be sure, I tested the PC-286VF with Gotek directly attached to it, but the save problem was the same.

Is it a problem of "AT32F415" or a problem of FDI image?

By the way, are there any plans to support D88?