keirf/flashfloppy #933: Archimedes 3010 and ADFs under 800kB?

micv2, 2024-09-14:
Hi, 
I have an A3010 with Gotek & latest FlashFloppy. How can I play the game Pack-Mania with an ADF of 795kB? 
I get Error 31. Other ADFs with 800kB works very well 👍

Best Regards,
MiC

keirf, 2024-09-14:
Easiest would be to pad it to 800kB. On Linux or similar this would be easy: `truncate -s 800K name.adf`

micv2, 2024-09-14:
Thank you Sir, Windows can't do that? 😉

keirf, 2024-09-15:
You will have to research it yourself for Windows. Or perhaps use Windows Subsystem for Linux (WSL)?

micv2, 2024-09-15:
Hi, ok thank you, it can be closed.

Regards,
MiC

micv2, 2024-09-16:
Hello keirf, 

on Sunday I got about 30 (too small) ADFs to the required 800k using Linux. Now (after work) I sat down at Archimedes (a3010) and wanted to play Pacmania. The game doesn't work. The ADF now has 800k, GOTEK shows no error, but Archimedes shows errors for every converted ADF, why? 

Other 800k or 1600k games work perfectly. 
Sir, it would be nice if you could help me again. I'm attaching the files here. 

Thank you very much 👍

[test.zip](https://github.com/user-attachments/files/17019691/test.zip)

[image]

keirf, 2024-09-16:
I assumed they were simply truncated, but perhaps these smaller ADFs are not so simple to convert to 800kB form. Do they work in full emulation on a PC? Anyway, your best bet is to ask on an Acorn-specific forum such as stardot.org.uk. I am sure someone there will be able to tell you what to do with these image files!

keirf, 2024-09-16:
Oh, actually now I see that you actually created a completely empty 800kB ADF. It is full of zeroes!

Please try the attached, correctly padded, ADF
[pacmania.zip](https://github.com/user-attachments/files/17019886/pacmania.zip)

micv2, 2024-09-16:
Ok, I'll try. But is my FF.CFG set correctly? (.zip) I have the jumper on JC&S1.

micv2, 2024-09-16:
"Oh, actually now I see that you actually created a completely empty 800kB ADF. It is full of zeroes!"

how could this happen? 🤦‍♂️

keirf, 2024-09-16:
The only non default option you need in FF.CFG is `host = acorn`

micv2, 2024-09-16:
yes Sir, the game is running 🎈🎈

I used Cygwin under Win10 to convert, it failed..😟

keirf, 2024-09-16:
All I did on Linux was `truncate -s 800K pacmania.adf` where `pacmania.adf` must be the name of your existing ADF image. Otherwise it will probably create a new empty 800kB file!

closed 2024-09-16 by keirf

micv2, 2024-09-16:
Ok, thanks Sir, 🙂