keirf/flashfloppy #934: SD Card Detected Signal

SukkoPera, 2024-09-16:
Quoting the _Enhanced Gotek_ wiki page:

> Card Detect should float when a card is inserted, and be driven low when no card is present. For simplest connection, this will require a card-detection switch which is open when a card is inserted, and closed when ejected. A switch with the opposite sense will require extra circuitry (eg. pull-up resistor and an open-drain MOSFET to invert the switch signal).

Is this a hardware limitation or could that be made configurable in the software?

keirf, 2024-09-16:
It could be configurable but it is quite easy to fix up in hardware so...

SukkoPera, 2024-09-16:
I see, but then I think the only supported behavior should be the other one. The vast majority (meaning actually all of them) of the SD slots I've seen short two pins when the card is _inserted_, and that makes it impossible to achieve the currently expected behavior without extra components.

keirf, 2024-09-16:
But I can't now change the default behaviour. Perhaps we could allocate a new identifier in pins PC15:12

SukkoPera, 2024-09-16:
OK, never mind. Can you just tell me if the following circuit is correct?

[image]

JP1 is a jumper that allows selecting the normal or inverted behavior.

keirf, 2024-09-16:
Looks about right to me.

closed 2024-09-17 by SukkoPera