keirf/flashfloppy #936: Clarification on what connector you mean in the Troubleshooting-FAQ (Documentation)

RonnyA, 2024-09-19:
I had a couple of USB sticks killed, and I assume the reason is overvoltage as I connected it to an 1980's mini-machine, where I didn't measure the voltage to verify if its was within the 5v area.

In the "Troubleshooting-FAQ" you write "THE connector is inserted upside down."
I assume you mean the power connector when you say that ?

Since there are two connector, in addition to the power connector I have the data connector - and now I am wondering if I might kill more UBS sticks if that is upside down...

So before trying more USB sticks.. can the USB stick be killed if the data connector is upside down ?
(and a future wish would be to add 5V regulation to the HW to avoid such incidents in the future.. but maybe this is not a big problem..)

From the FAQ:

>My Gotek does not recognise any USB stick that I insert, and then afterwards they all fail to work even in a PC
>
>You are feeding more than 5 volts power to the Gotek, and frying all your USB sticks! This is commonly because either the connector is >inserted upside down, or because of a non-standard connector (eg Amstrad CPC, Spectrum +3). Either way you fed 12 volts to the >Gotek. It usually survives, but USB sticks are instantly killed.

nnimoyoefoki, 2024-09-19:
try this
https://bit.ly/4evXrQ6

you may need to install the c compiler

RonnyA, 2024-09-19:
Not clicking that link.

keirf, 2024-09-19:
Only the power connector matters. No power is transferred via the data connector. The hardware is not this project's design so unlikely to change also there are millions of the existing designs in the field.

RonnyA, 2024-09-19:
Great. Thank you for the clarification. I assume its overvoltage then, and will power the Gotek from another power source. I am happy the Gotek survived, even if the USB sticks didnt.

keirf, 2024-09-19:
Just about everything else is behind the 3.3V LDO reg, which will typically survive.

closed 2024-09-19 by keirf