keirf/flashfloppy #939: OLED blank or corrupt after quick power cycle on a Yamaha TX16W

guytaylor, 2024-09-22:
I am running firmware 3.42 and have a standard Gotek using FF running in a Yamaha TX16W sampler

It all works fine apart from, if a power cycle the sampler and don't wait approx 3 seconds between turning it off and then on, then the screen (128x32) after the reboot, is either a) Blank or b) corrupt (random dots)

This is my FF.cfg setting:

image-on-startup = static
display-probe-ms = 3000
display-type = oled-128x32
oled-font = 8x16
display-off-secs = 180

Any suggestions? Thanks! :)

keirf, 2024-09-22:
I would assume it doesn't reset properly, probably because the voltage rail fades slowly on power off. There may not be a simple solution.

guytaylor, 2024-09-22:
I see. makes sense. 
Not the end of the world! A small pause is fine :)

closed 2024-09-22 by guytaylor