keirf/flashfloppy #940: Regarding the Size of Image Files

bodong001, 2024-09-24:
How to achieve a legitimate 2.88M image file on the emulator? I changed the file format to 2.88M in the WinImage tool and obtained an image file, but the emulator does not support it!, May I ask how to obtain a larger image file?

keirf, 2024-09-24:
What is the exact byte length of your 2.88M IMG file? Can you share it?

WinImage can do many other image types than floppy images, and some of those have IMG extension too!

In short, 2.88M IMG floppy image file should "just work", at least with out-of-the-box FlashFloppy configuration.

philpem, 2024-10-02:
Another possibility is the disk might have bad sectors. If the format wasn't specified on the command line, the bad sectors might have been skipped.

philpem, 2024-10-02:
Are you creating the image from scratch?
Did you save the image as a .IMA file, or a .IMZ?

.IMZ is compressed and won't be readable by Flashfloppy. You need .IMA, which is a raw sector image.

You should also configure Flashfloppy (in FF.CFG) - the details are in the default, but you want PC style drive interfacing and the host machine type set to PC. That configures how Flashfloppy interprets raw sector images like IMA files.

closed 2024-12-03 by keirf