keirf/flashfloppy #944: Akai S950 DS/HD floppy images

chickeneps, 2024-09-30:
I am having issues with the S950 and HD images. The S950 supports both DD and HD floppies. The Wiki says nothing about it, though it mentions for the S950 "interface = akai-s950" and for the predecessor S900 "interface = shugart".

A DD floppy image is 819200 bytes, thus a HD floppy image would be 1638400 bytes. So I create a blank file 1638400 bytes large and fill it with 0xCC for the checks below.

What happens on the S950 is that if you format that image, it formats fine. The Track in the OLED goes from 0-79, like normal.

But then, when you try to save something, the S950 says "OOPS Disk is unprepared you must FORMAT IT". Now, whats interesting is that when that HD image is formatted, nothing is erased on the image - the contents are still 0xCC (see above). Whereas when you format a DD image, all contents are erased and the new file system is entered.

Here's an article concerning this, but it talks in HxC terms, not FlashFloppy. https://www.donsolaris.com/?p=2317

So I'm after a way to have DD images and HD images on the same USB drive, and both fully functional - similar to how the Ensoniq ASR-10 works. I have a feeling the DD images were seen and supported when S950 support came in, but not HD images.

keirf, 2024-09-30:
The problem will be that the s950 thinks it's a DD disk and formats it DD. Of course FF sees no HD sectors and so drops the writes on the floor. And Akai format clearly doesn't verify as it goes.

Density signalling is via pin 2. At least usually. I can't remember if s950 is a device with variant drive pinouts.

Please attach the exact FF.CFG file you have on your usb stick.

chickeneps, 2024-09-30:
## FF.CFG: Ensoniq EPS/EPSm FlashFloppy Configuration File

# Place in the root folder or FF/ subfolder of your USB drive.
# Uncommented lines below are the default settings.
# Uncommented options cannot be overridden by settings in other config files.

##
## DRIVE EMULATION

# Floppy-drive interface mode
# shugart: P2=DSKCHG, P34=RDY
# ibmpc: P2=unused, P34=DSKCHG
# ibmpc-hdout: P2=HD_OUT, P34=DSKCHG
# akai-s950: P2=HD_OUT, P34=RDY (Akai S950)
# amiga: P2=DSKCHG, P34=DRIVE_ID (not generally needed: prefer 'shugart')
# jc: JC closed: ibmpc, JC open: shugart
interface = akai-s950

# Host platform: Improves image-format detection for generic types such as IMG
# acorn: Acorn ADFS
# akai: Akai (S01, S20, S950)
# dec: DEC (RX33, RX50)
# ensoniq: Ensoniq (ASR, TS, etc)
# fluke: Fluke 9100
# gem: General Music (S2, S3)
# kaypro: Kaypro
# memotech: Memotech
# msx: MSX
# pc98: NEC PC-98
# pc-dos: PC DOS Format (geometry determined from Bios Parameter Block)
# tandy-coco: Tandy Color Computer (CoCo)
# ti99: TI-99/4A
# uknc: UKNC / DVK Soviet PDP-11
# unspecified: Common default geometries (including IBM PC)
host = akai

# Pins 2 & 34 manual configuration
# auto: Auto-configure from interface= setting
# nc: Unused/No Connection
# low: Always 0 volts (0v)
# high: Always 5 volts (5v)
# rdy: Drive ready (Ready = 0v)
# nrdy: Logical complement of above
# dens: Density mode (High Density = 0v)
# ndens: Logical complement of above
# chg: Disk changed (Changed = 0v)
# nchg: Logical complement of above
# Values: auto, nc, low, high, rdy, nrdy, dens, ndens, chg, nchg
pin02 = nrdy
pin34 = auto

# Forcibly write-protect images, or respect the FAT read-only attribute?
# Values: yes | no
write-protect = no

# Filter glitches in the SIDE-select signal shorter than N microseconds
# Values: 0 <= N <= 255
side-select-glitch-filter = 0

# Rotational offset of data after a track change
# instant: No rotation during track change
# realtime: Emulate rotation of disk while track is changing
# Values: instant | realtime
track-change = instant

# Index pulses suppressed when RDATA and WDATA inactive?
# Values: yes | no
index-suppression = yes

##
## STARTUP / INITIALISATION

# Disk image loaded or ejected at startup?
# Values: yes | no
ejected-on-startup = no

# Which image (or folder) is selected at startup?
# last: Last-selected item at power-off (recorded in IMAGE_A.CFG)
# static: Static path specified in IMAGE_A.CFG
# init: First item in root folder
# Values: last | static | init
image-on-startup = last

# Time in milliseconds to attempt to probe attached display.
# You may set this to 0 if you have a 2-digit LED display attached.
# Values: 0 <= N <= 65535
display-probe-ms = 3000

##
## IMAGE NAVIGATION

# Auto-select the current file after N seconds
# N=0: disable auto-select
# Values: 0 <= N <= 255
autoselect-file-secs = 1

# Auto-select the current folder after N seconds
# N=0: disable auto-select
# Values: 0 <= N <= 255
autoselect-folder-secs = 1

# Navigation mode for selecting images/slots
# native:  Navigate through all valid images/dirs
# indexed: Navigate through DSKA0000, DSKA0001, ...
# default: native unless overridden by HxC-compat-mode config
nav-mode = native

# When navigating slots or folder, loop at min/max?
# Values: yes | no
nav-loop = yes

# Actions of first two buttons.
#              B1     | B2     | Both
# zero:        Prev   | Next   | Slot 0
# eject:       Prev   | Next   | Eject/Insert
# rotary:      Up-dir | Select | -
# rotary-fast: Prev   | Next   | Up-dir [Prev/Next are accelerated]
twobutton-action = zero

# Type of rotary encoder connected to pins PC10 and PC11, identified by
# fraction of a Gray-code cycle performed per detent/click.
# If default value ('full') requires multiple clicks/detents to move position
# then change to 'half' (if 2 clicks per move) or 'quarter' (if 4 clicks).
# Values: none | quarter | half | full
rotary = full

##
## DISPLAY

# Display Type: <type>[-rotate][-narrow][-sh1106]
# auto: Auto-detect (7-seg LED, LCD, OLED)
# lcd-NNx02: NNx2 backlit LCD with I2C backpack (16 <= NN <= 40)
# oled-128xNN: 128xNN I2C OLED (NN = 32 | 64)
#  -rotate: OLED view is rotated 180 degrees
#  -narrow: OLED view is restricted to Gotek display cutout
#  -sh1106: OLED controller is SH1106, not SSD1306
# Values: auto | lcd-NNx02 | oled-128xNN[-rotate][-narrow][-sh1106]
display-type = oled-128x64-rotate

# OLED Font. Narrow and wide options.
# Narrower 6x13 font permits:
#  - More characters per row
#  - Use of Gotek display cutout (with "display-type=oled-128x32-narrow")
# Values: 6x13 | 8x16
oled-font = 8x16

# Turn an LCD or OLED display off after N seconds of inactivity
# N=0: always off; N=255: always on
# Values: 0 <= N <= 255
display-off-secs = 60

# Automatically switch LCD/OLED display on when there is drive activity?
# Values: yes | no
display-on-activity = yes

# LCD/OLED long filename scroll rate in milliseconds per update
# Values: 100 <= N <= 65535
display-scroll-rate = 200

# LCD/OLED pause time at start/end of scroll, in milliseconds
# Zero means endless scroll
# Values: 0 <= N <= 65535
display-scroll-pause = 2000

# LCD/OLED long filename scroll rate during navigation (ms per update)
# Values: 0 <= N <= 65535
nav-scroll-rate = 80

# LCD/OLED long filename pause before scroll, during navigation (milliseconds)
# Values: 0 <= N <= 65535
nav-scroll-pause = 300

##
## MISCELLANEOUS

# Speaker volume for head STEP
# Values: 0 <= N <= 20
step-volume = 10

# Report the specified version number to host software
# Values: <quoted-string> ("" means report real version)
# eg. da-report-version = "v3.0.0.0"
da-report-version = ""

# Automatically extend certain types of truncated image file (SSD,DSD,TRD)?
# Values: yes | no
extend-image = yes

chickeneps, 2024-09-30:
The interface and host are what is suggested in Host Platforms.

chickeneps, 2024-09-30:
pin02=dens seems to do the trick. Can you add that to the Host Platforms note?

keirf, 2024-09-30:
pin02 and pin34 are pin-specific overrides for interface. You can just delete those and then reset cached config, or set them =auto which means do what interface= says. I think you've been bit by this before.

closed 2024-09-30 by keirf

chickeneps, 2024-09-30:
Bit? =) More like being eaten day after day after day =) So, unless mentioned otherwise, both settings should always be on "auto"?