keirf/flashfloppy #945: Problem of head section on Thomson computer

Neo2003, 2024-10-04:
Hello,
Playing with flashfloppy on Thomson computer, I was able to convert sap images to edsk images and use all the 1 sided 320ko floppy images. Thomson are using 16 sectors of 256 bytes, 80 tracks.
This computer also supports double-sided floppy disk, but it does use the floppy as 2 independent sides. So it reads side 0, then side 1.
The Thomson controller is a THMFC1.
My problems begins when I switch to 2 sided images. Whatever I create a edsk with 2 sides or just create a mapping to the images, it does not work. The Gotek switch to side 1 while the computer still expect to see only side 0, then it crashes.
2 sides edsk or this definition lead to same problem. fd files are just stupid sector contents of full side 0, then full side 1 sequentially.
```
[fd::655360]
file-layout = sequential
cyls = 80
heads = 2
interleave = 7
secs = 16
bps = 256
gap3 = 27
tracks = 0-79.0
  id = 1
tracks = 0-79.1
  id = 1
```
Instead of reading 0.0, 1.0... the Gotek switches to 0.1 after 0.0 while it should not.
The floppy disk drive used on this computer is this one 
https://bitsavers.trailing-edge.com/pdf/sony/floppy/9-975-176-01_MP-F53W_Product_Specifications_Sep85.pdf

Is this paper it is stated:
`3-3-4. HEAD SELECT
A true (low) level on this input will cause Head 1 (upper) to be selected. A false (high) level on this input will cause Head 0 (lower) to be selected. If the HEAD SELECT signal changes during either a write or erase operation, the head will not change until both the ERASE GATE and the WRITE GATE signal become false (high).`

So can you please guide me on how to make the gotek not switch to side 1 when it should not.

Neo2003, 2024-10-04:
Found a workaround by forcing to go back to head 0 if on head 1

```
[fd2::655360]
file-layout = sequential
cyls = 80
heads = 2
interleave = 7
secs = 16
bps = 256
gap3 = 27
tracks = 0-79.1
  h = 0
```
But trying to do the same with dsk does not work like if it was ignored.

keirf, 2024-10-04:
CPC DSK files contain the sector headers so don't use the IMG.CFG definitions. Those definitions are only used for raw sector images (eg. IMG).

Neo2003, 2024-10-06:
Ok, so maybe correct the IMG.CFG file. 
 1. <tagname> matches images of the form *.<tagname>.{img,ima,dsk}

I am still search why it moves to head 1 after a track on head 0 is read, when I make a dske with Thomson TO8 disk or when I use a .fd with the IMG.CFG without h=0

closed 2024-12-03 by keirf