keirf/flashfloppy #946: .img files not read on 435

chrisvagnozzi, 2024-10-06:
AOpen AX34, P3 motherboard. Attempting to load DOS 6.22 and Win95/98 .img files. None are working. Jumper closed on S1, tried also S1+JC. Gotek is reading images showing yellow light and doing nothing. Twisted floppy cable, assigned 1.44MB A:\ in BIOS. Tried with just .img file, tried with .img file and FF/FF.CFG with ibmpc and pc-dos settings.

keirf, 2024-10-06:
Have you tried another drive, Gotek or real? The setup you describe should work fine. So long as the IMG file is exactly 1.44mb. you don't need any FF.CFG so start off by deleting that and resetting any cached config (hold buttons for a few seconds with no stick inserted until the display changes). Jumpers at S1 and JC should suffice. Make sure floppy is enabled and configured correctly in bios.

chrisvagnozzi, 2024-10-06:
I popped my regular drive in and found it wasn't working either. Turns out, my ESS ISA sound card has an IDE header that seemed to be causing an IRQ conflict with the FDC. I removed the card and physical drive works. (I'll have to disable the IDE in the ESS.CFG) Anyway, back to the Gotek, the device now reads my Win95+Win98 boot disks, but not the DOS 6.11 disk (3-set disk). Trying to boot off the DOS 6.22 Disk 1 results in nothing. Is there additional configuration required for this, perhaps setting something up in FF.CFG on the USB drive? I'm using S1+JC, just to extra confirm. Thanks Keirf!

keirf, 2024-10-06:
No, a 1.44mb IMG is a 1.44mb IMG as far as FlashFloppy is concerned. That's so long as you didn't put host=pc-dos in FF.CFG. Be sure you have removed FF.CFG *and* reset the cached config inside the Gotek. With that and IMG files exactly 1.44mb in size (to the byte) and S1+JC you should be good.

chrisvagnozzi, 2024-10-07:
Please consider this resolved. Resetting the cached config helped. Also learned that ejecting usb drives safely in Win10/11 prevents issues with certain .img files not loading cleanly. Thank you for the timely help!

closed 2024-10-07 by keirf