keirf/flashfloppy #949: 360K 5-1/4" Floppy imulation

ronf3-1964, 2024-10-23:
I am trying to get a 360K 5-1/4" floppy drive replaced in an old CNC control. The disk are standard DOS format. I have a GOTEK USB drive. I have flashed the firmware on the unit. I cannot seem to figure our how to create the USB stick with the files. Can someone please provide me with the instructions on how to do that.

Thank you

keirf, 2024-10-23:
Use software such as winimage or rob Smith's diskflashback to mount an IMG file in windows and populate it with files. FlashFloppy expects IMG files on a FAT32 formatted usb drive.

ronf3-1964, 2024-10-23:
I tried winimage but it does not support 360K.

keirf, 2024-10-23:
Share the files and I can probably make you one.