keirf/flashfloppy #952: Eject menu confusion / write protection

xsrf, 2024-11-16:
As a new user to the software (and after fitting an OLED) I'm pretty confused by the eject menu. Because of this, it took me like 2h to debug and find out just how to disable Write-Protection for an image properly even after reading the documentation available.

When I found out that all my img-Files placed on the usb drive are mounted write protected, I search the documentation and found the `write-protect` option in https://github.com/keirf/flashfloppy/wiki/FF.CFG-Configuration-File
It told me I had to press the eject-Button for 2s.
Since there is no eject button on the stock gotek, I configured `twobutton-action = eject`. Pressing both buttons for 2s didn't do what I wanted but opened the eject menu instead where I could toggle write protection. I then used `Exit to selector` and the floppy was still write protected. I opened the menu again, and sure, write protection showed on again.

I further searched docs and Issues and found https://github.com/keirf/flashfloppy/issues/135 and https://github.com/keirf/flashfloppy/issues/516 which told me a) that write protection flag won't be persisted for the image (fine) and b) some USB sticks might report as write protected.
Since all my images show up write protected and toggling write protection for me always instantly reverted the flag, I assumed my USB stick might report as read-only and I tried `usb-ignore-wp = yes`. This didn't help of course because it never got implemented in the regular release, but I didn't know that. 

After more tinkering I accidentally found out that write protection was successfully kept disabled when exiting the menu on its title screen `**Eject Menu**`. Later I found out that `Exit & re-insert` would have actually been the right option despite mentioning the action of ejecting and inserting the disk again, which I thought would reset the write protection for sure.
Also, entering the menu again would reset the Write Protection to ON immediately so I can't lookup the status via the menu, I have to confirm by writing files.

To me, the menu items were confusing...

Opening the `**Eject Menu**` ejects the disk. Didn't expect that, but fine. Guess the name gives it away.

`Exit to selector` resets the write protection flag and re-inserts the disk.
`Exit & re-insert` keeps the write protection flag and re-inserts the disk.

I get that, in theory, `Exit to selector` brings you to `Navigation/Selector` mode where no disk is emulated, but in reality (from my experience with default settings) it just mounts the selected disk again (because of auto-mount after 2s). So it's not really different from re-inserting the disk.
And since the selector menu is the menu I'm coming from when opening the eject menu, returning to selector kinda implied to me that it won't do anything to the disk.

I also get that the disk is already "ejected" in eject menu, so "Exit & re-insert" is the right term in this analogy - but it's confusing.

So I suggest at least documenting this in the Wiki as a new Page "Menu system" under "Usage". Here is my suggestion, but be aware that I know basically nothing and just started using the software 2 days ago:

## Eject Menu
When used with display or OSD, flashfloppy provides an optional "Eject Menu" that can be accessed by pressing e.g. up+down together or an installed rotary encoder (see `twobutton-action` at https://github.com/keirf/flashfloppy/wiki/FF.CFG-Configuration-File).
Opening the "Eject menu" instantly ejects the disk - it's not readable by the host system anymore.
You can navigate through the menu items by pressing up or down or using the rotary encoder. Up+down or pressing the rotary encoder executes the selected item.

The available Menu Items are:
- **Write Protection: ON/OFF**: Toggles write protection for the currently selected image. This is only temporary until another image is selected or the disk is ejected. Use `Exit & re-insert` afterwards!
- **Copy**: selects the current image for paste operation - won't do anything immediately
- **Paste**: Pastes a copy of the via "Copy" selected image into the current folder. A "0001" etc. suffix might be added if an image with the same name already exists. This operation asks for additional confirmation.
- **Delete**: Deletes the currently selected image. This operation asks for confirmation.
- **Exit to selector**: Exists to the selector menu with no image currently mounted (disk ejected). By default, auto-mount will however re-mount the selected image again within 2s
- **Exit & re-insert**: Exits and re-mounts the selected image instantly. Use this option to keep the Write protection flag set before.
- **\*\*Eject Menu\*\***: same as Exit & re-insert

keirf, 2024-11-16:
Thanks, I have edited your text and inserted it into the wiki. It's not the case that ejecting loses the current write-protect status. I have just tested that myself, and therefore removed that from the wiki text.

As for your broader problem, really there are two better solutions than requiring a new user to understand the semantic nuances of FlashFloppy's menus and operating modes:
1. The write-protect toggle should be persisted to USB drive (#516)
2. All the files on your USB drive must have the FAT read-only attribute set. I don't know why: Perhaps inherited from the read-only attribute of the whole drive or folder they reside in? The quickest fix to make them all writeable will be using `chmod` (Linux/Mac) or `ATTR` (Windows/DOS) in a command-line shell.

closed 2024-11-16 by keirf

xsrf, 2024-11-16:
So, I fixed the issue that all images are always write-protected. I added `write-protect = no` explicitly to `FF.CFG` and that fixed it. I then removed the option again and it still uses the read-only attribute.
I've used `write-protect = yes` yesterday but removed it again. Is it possible it got cached as new default?

However, opening Eject Menu still resets the flag for me instantly.

xsrf, 2024-11-16:
> I've used write-protect = yes yesterday but removed it again. Is it possible it got cached as new default?

Ah shit, that's actually documented :/ missed that. My fault, sorry.

keirf, 2024-11-16:
Yeah the caching of FF CFG items confuses new users greatly. Some config items want to be active even before/without a stick inserted so the caching behaviour does broadly make some sense. It is how it is!

keirf, 2024-11-16:
The fact that opening the eject menu resets the WP toggle is really weird... Are you in direct navigation mode not using autoboot.hfe file selector?

I think I need to persist the WP toggle for next release anyway. Having tried it out myself today, it's definitely too ephemeral right now, and confusing (and annoying) af.

keirf, 2024-11-16:
I wonder if it's related to going into the eject mode via two buttons. I can imagine that might slip back into navigation mode for a split second, as you can never press both buttons exactly simultaneously...

Anyhow if I make WP persist this issue goes away naturally. So I should just do that.

xsrf, 2024-11-16:
It might be that I use two buttons, yes. I really tried hard to hit them together but most of the time it obviously navigates before entering the eject menu. But since the right  disk is still selected, I figured that wouldn't be an issue. But if just entering navigation mode resets the flag, then yes, that might be the cause.

But it's not a real issue and with persisting WP flag it should be fixed too ;) Thx