keirf/flashfloppy #953: Fresh 3.42 flash on STM32F105 presents error E3

53hornet, 2024-12-03:
Hello. I am troubleshooting my Gotek emulator with the STM32F105 chip. I just flashed the latest version (3.42) of FlashFloppy using the Windows instructions. The download and verify appeared successful and on boot the drive displays `FF` in the two digit display. 

I'm having trouble actually getting the drive to recognize images however. I tried several .IMG files and three different flash drives so I think I'm configuring something wrong. I wrote an FF.CFG for my IBM PC compatibles like this:

```
interface = ibmpc
host = pc-dos
```

All three USB drives have image files on them but all of them just read `E3` on the two-digit display. I saw a closed issue on this which mentioned the possible drive error codes like an empty or unformatted drive but that doesn't seem to be the case here. On two of my USB sticks, the display flashes between `00` and `E3`. 

Does anyone have any guidance on where I should look next to find out more? Thanks!

keirf, 2024-12-03:
Revert to `host = unspecified` as it is usual for the disk type to be inferred from the IMG file size.

What exact size, in bytes, are the IMG files?

53hornet, 2024-12-03:
```
% ls -l *.img *.IMG
-rwxr-xr-x    1 root     root        368640 Aug 18  2011 68X2529.IMG
-rwxr-xr-x    1 root     root        368640 Feb 12  2005 ATDG207.IMG
-rwxr-xr-x    1 root     root       1474560 Nov 28  2014 Disk1.img
-rwxr-xr-x    1 root     root       1474560 Nov 28  2014 Disk2.img
-rwxr-xr-x    1 root     root       1474560 Nov 28  2014 Disk3.img
-rwxr-xr-x    1 root     root       1474560 Dec 15  2014 Suppdisk.img
```

Here are all of the image files on disk and their exact sizes. They should be a combo of 360k and 1.44m images. Using a `host = unspecified` config in FF.CFG yields the same result.

keirf, 2024-12-03:
Try flashing the logging firmware (detailed on a wiki page) and collect an FFLOG.TXT file. That will determine the full error code as it is truncated on the two digit display. Those files should indeed work.

53hornet, 2024-12-04:
I tried updating to the logging firmware and ended up getting the same error. So I went a reflashed the drive from the beginning using the same PC, defuse, and firmware release as before. This time it recognized the drive and let me update the firmware to logging, recognizing the USB drive. I reflashed back to the non-logging firmware and now it's reading disk images in my IBM. 

For now I'll close this indicating it must have been a bad flash/reprogramming procedure on my part. But FF is working great now. Thanks for all the help!

closed 2024-12-04 by 53hornet