keirf/flashfloppy #955: Strange behaviour of CD pin on OpenFlops board

Squirrel61, 2024-12-16:
Recently, I've built two OpenFlops boards. These boards have a header for an SD-card, so I connected one to see how it works compared to USB stick (it seems more convenient). However, I only got it working after disconnecting the CD (Card Detect) pin. It seems that the CD logic is reversed, because on all my SD-card modules, the CD signal is low (contact closed) when a card is inserted where FlashFloppy is expecting a high signal when a card is inserted.

What exactly does FlashFloppy do with the CD pin signal? Is there any functionality connected to it, or can I just leave the pin open/floating so the SD-card works? And is it possible to reverse the FlashFloppy detection of the CD pin, or make it configurable (I don't know an SD-card module that closes the CD contact when the card is ejected, all of them close the contact when the card is inserted).

keirf, 2024-12-16:
If you don't need SD hotplug you can leave it open.

Squirrel61, 2024-12-16:
Ok thanks. But if I do need the hotplug function, how can I make it work? I can't find an SD-card module with reverse logic so that means I have to find a way to invert the signal in hardware?

keirf, 2024-12-16:
Yeah it's not currently software configurable. You would need a transistor (eg 2n3904). CD pin to base, Ground to emitter, CD output to collector.

Squirrel61, 2024-12-16:
Thank you, will try that solution.

closed 2024-12-16 by Squirrel61

keirf, 2024-12-16:
Oh, also a pull-up at the base. Anything around 1k to 10kohm, to 3v3 or 5v rail.