keirf/flashfloppy #961: Slow Roland S-50 System load

weeksd59, 2025-01-06:
Hi

FlashFloppy v3.42
Gotek SFRKC30.AT3 with OLED
Artery AT32F415
Roland S50 Sampler

There is a 25 second delay before data starts loading into my S50 from the Gotek.

When I boot the S50 up, the S50 screen shows "Please Insert system disk". So I put the USB stick into the emulator and the S50 immediately displays "S-50 System Ver. 2.0". So far, so good. 

What I think is happening next is that the S50 System software is being loaded but very slowly. The Gotek track display gradually progresses through to track 7.1 taking about 25 seconds. Then, the sound data is loaded at normal, expected speed.

According to the S50 service notes, the expected action is as follows:-

_"On a read or write command from the CPU, the Floppy Disk Controller (FDC) pulls MOTOR ON low to let the FDD (Floppy Disk Drive) start the motor. 

When the motor running has reached stable condition, the FDD signals the CPU through 1/0 gate with a low READY. 

The low READY allows the CPU to issue a command which enables reading or writing to/from the disk.

In the read mode, the FDC reads data from FDD in serial format and sends it to the CPU in parallel 8 bits."_

Can you offer any help please?

keirf, 2025-01-06:
https://github.com/keirf/flashfloppy/wiki/Host-Platforms#slow-loading

weeksd59, 2025-01-06:
Hi Keir

Thanks for the quick reply.

I've used the suggested IMG.CFG file and adjusted the "tracks" setting
accordingly which has solved the problem.

I'm still in debug mode and notice the log contains some repeated errors.

Are you able to determine what the problem might be please (see attached).

Many thanks
Dave

> https://github.com/keirf/flashfloppy/wiki/Host-Platforms#slow-loading

-- 
Cheers
Dave

** FlashFloppy 3.42
** Keir Fraser ***@***.***>
** github:keirf/flashfloppy

Build: Jan 11 2024 13:54:17
Board: Standard
Config: Flash Slot 0 (ver 2, size 74)
Interface: Shugart (pin2=chg, pin34=rdy)
I2C: OLED found at 0x3c
OLED: SSD1306
Display: LCD/OLED

> USBH_USR_Init
I2C: Error (0400)
I2C: Error (0400)
I2C: Error (0400)
I2C: Error (0400)
I2C: Error (0400)
I2C: Error (0400)
I2C: Error (0400)
I2C: Error (0400)
I2C: Error (0400)
I2C: Error (0400)
> USBH_USR_ResetDevice
> USBH_USR_DeviceAttached
> USBH_USR_ResetDevice
> USBH_USR_DeviceSpeedDetected
> Device speed: Full
> USBH_USR_DeviceDescAvailable
 VID : 08EC
 PID : 0008
> USBH_USR_DeviceAddressAssigned
> USBH_USR_ConfigurationDescAvailable
> Class connected: 08 (MSC)
 Manufacturer : Kingston
 Product : DataTraveler 2.0
 Serial Number : 0DE19A5100C328BD
> USBH_USR_EnumerationDone
> USBH_USR_UserInput
Cache 42 items
Config: Erased Slot 0
Config: Written to Flash Slot 1
Mode: Native
Cache 42 items
Current slot: 0/2
Name: '_S-50_SYSTEM_BOOT_DISK_VER__2_0.os' Type: img
Attr: 20 Clus: 00000013 Size: 737280

weeksd59, 2025-01-06:
> I2C Error is a problem talking to the OLED display. They aren't very good,
> especially at power on. If you only see errors at startup, don't worry
> about it.

Yeah, everything else works fine.

I'll turn debug off and get using it.

Thanks ever so much for the fast replies and expert solutions

Regards
Dave

closed 2025-01-07 by keirf