keirf/flashfloppy #968: Add a new configuration to pin2 to be driven by an IMG.CFG disk defintion

s100projects, 2025-03-02:
In the case of 8" drives, 2-sided drives ground pin #10 (-TWO_SIDED) to indicate when a double-sided disk is inserted.  

When a double sided image is selected in FlashFloppy, but the switch is not in the correct state, the operating system will report sector errors (because it thinks it is dealing with a single sided disk image).  The -TWO_SIDED switching  is usually handled by an external 50-pin to 34-pin adapter, but if that adaptor is housed inside a system cabinet, it can be difficult to easily or quickly switch between single and double-sided images.

Control of such a pin-2 configuration could be included in the IMG.CFG file disk definitions as so:

Disk Definition for a DSDD CompuPro Image:
> [::1251840]
> cyls = 77
> heads = 2
> interleave = 1
> rpm = 360
> gap3 = 42
> tracks = 0.0    ## Boot cylinder, head 0
> pin02 = low   ## Double sided disk image
> ...
 
This would allow disk definitions to automatically indicate the correct cylinder configuration to the 8" controller.

s100projects, 2025-03-02:
FYI: I have AT32F415 based Goteks with the TFT panel and selector knobs.

keirf, 2025-04-19:
This is a good idea. Let me think.

s100projects, 2025-05-12:
Thanks!  Let me know if you need any additional details from me.

Best Regards,

Jeff