keirf/flashfloppy #971: indexed-prefix Not Working as expected

fraser125, 2025-03-04:
tl;dr; Adding 1 char to the length of the first part of the filename causes E34 and is unusable. 
I was attempting to prefix the image files with "RolandMC50_" so I knew the source when viewing the image names. I found that RMC50_ worked and then I gradually increased the length until the steps were reproducible. 

Additional Details may be available: [GitHub Discussion](https://github.com/keirf/flashfloppy/discussions/969)

### Working Example (Successfully formatted images with the Roland MC-50)
.\FF\FF.CFG 
```
nav-mode = indexed  
indexed-prefix = "RoMC50_"
```

.\FF\IMG.CFG taken from [Roland IMG.CFG in this repo](https://github.com/keirf/flashfloppy/blob/master/examples/Host/Roland/IMG.CFG)

2 image Files in the root of the USB Flash Drive 
RoMC50_0000.mc.img
RoMC50_0001.mc.img

### Failing Example E34
**Only change these things (Case Insensitive)**
indexed-prefix = "RoLMC50_"`
**Rename files to match**
RoLMC50_0000.mc.img
RoLMC50_0001.mc.img

This makes the IMG.CFG line 1 character longer and the filenames 1 character longer. I believe one of these 2 are causing the E34 error.
I really lean towards the Filenames being to long because 
The first block of the name goes from 12 -> 13 characters. ex. RoMC50_0001 (12) -> RoLMC50_0001 (13)
I tried to find the possible code areas for this issue, but C is not a strong language for me.

Obviously not a breaking bug and easy to work around, but nice to get it fixed.

fraser125, 2025-03-25:
I'd like to confirm if my steps are reproducible. It's easy to work around this issue, but if it's not reproducible I could refine and verify the steps above.
Thanks!

keirf, 2025-03-25:
Sorry, yes I understand this issue having thought about it. The documentation needs updating to make it clear that the prefix string is limited to seven characters, and silently truncates to that length.

closed 2025-03-25 by keirf

keirf, 2025-03-25:
On second thoughts, I haven't documented the truncation. I may in future change the behaviour to raise an error instead. But the seven-character limit is now documented, including in the example FF.CFG file (pending next release of FF, which might be a while!).

commit cc9039c1 (2025-03-25): FF.CFG: Document indexed-prefix option string's seven-char limit

Fixes #971