keirf/flashfloppy #972: (Not a bug) Help requested with IBM PCjr setup

josh2112, 2025-03-24:
I have an IBM PCjr with a failing floppy drive, and I'm attempting to replace it with a Gotek. My Gotek has the AT32F435 chip and I've flashed it with 3..43. I have been unable to get any disk image to boot. I blame myself for not yet landing on the right hardware/.CFG configuration, not FlashFloppy.

Floppy cable:
 - I replaced the PCjr floppy cable with a standard 34-pin cable with the "twist".  I believe I have both connectors right-side-up (because putting them in any other configuration gives me a BIOS "ERROR H". Should I be using a straight-through cable?

Jumper settings:
 - I have the JC jumper disconnected, and the S1 jumper connected. I have read that for a PCjr the jumper should be set to S0, but that gives me Error H again.

In FF.CFG, I have the following values:
 - interface = ibm-pc
 - host = pc-dos

Floppy images:
 - I have a 180 KB DOS 2.10 image which appears to read, but fails with "Disk Boot failure".
 - I also found an image of the "PCjr sampler" which came with new PCjrs. It also fails with "Disk boot failure".
 - I have an image of Touchdown Football (one of my favorites as a kit because it used the speech adapter!). Upon boot, it slowly prints out "T".. "1".. "2".. "3".. then "Error 9".
 - Is it possible all my images are bad? They were downloaded, not created by me. Is there a test image I can try? (I'm not asking for copyrighted material, just any freely-available known-good floppy disk image that would allow me to confirm that the PCjr is successfully reading from FlashFloppy).

Thanks!

keirf, 2025-03-24:
You should use the same ribbon type as originally fitted. That's probably straight, no twist. With that you may find that jumper position S0 works. It may also affect orientation of the ribbon: On most systems FlashFloppy will actually report at power on whether the ribbon is wrong way round.

josh2112, 2025-03-25:
Thanks for the response! Non-twist cable is on the way.

Although I do think I'm reading data from the disk... I thought the "Disk Boot failure" message came from the PCjr BIOS, but looking at one of my disk images in a hex editor I found that string near the beginning. I changed it to some test value, saved the image and attempted to boot with it, and I got my changed string.

So I'm not so sure it's the cable, more likely my images are being read successfully and they're just not bootable for some reason.

keirf, 2025-03-25:
Looking at a PCjr schematic, I would expect S1 and straight cable to be correct. https://wiki.console5.com/tw/images/2/21/IBM-PCjr-Diskette-Drive-Adapter-Schematic-2.png

I would also expect JC connected vs disconnected to make no difference, but worth a try.

Is PCjr compatible with regular 360k DSDD PC compatible disks?

josh2112, 2025-03-25:
> Is PCjr compatible with regular 360k DSDD PC compatible disks?

That's right.

When the straight cable comes in tomorrow I'll cycle through all the jumper settings again.

josh2112, 2025-03-27:
Success! Straight-through ribbon cable, jumper S1, no jumper on JC, and interface = ibmpc is the way to go.

Thanks again for your help.

closed 2025-03-27 by josh2112

josh2112, 2025-03-27:
Interesting, I didn't consider that. I purchased the twist cable because this guy (https://youtu.be/j8vxedTU3-E?t=128) is using it (with S0 and JC). At any rate, my current setup works!