keirf/flashfloppy #980: IBM PC with FDC controller not detecting floppy inserted

troydm, 2025-05-02:
Hardware: Modded Artery AT32F415 based with KC30 Rotary Header and OLED display and Speaker
MB: MSI ms-7253 Socket AM2 based motherboard with builtin fdc controller that can support disks upto 2.88
FlashFloopy v3.44 firmware 
Drive is only detected when jumper in S1, floppy cable connected after twist, FF.cfg interface = ibmpc, putting jumper in S0 makes FDC controller totally ignore the drive

now only thing is working is when booting from floppy during pc startup and I see green led blinking and beeping in speaker of gotek, I can boot into ms-dos setup img on gotek, exit the setup with F3 and in command prompt dir A: and everything is fine, I can even run basic from floppy img

However if I boot into Windows XP or even dos 6.22 installed already on hdd I can't access floppy as if FDC controller is not initialized, in Windows device manager I see Standard Floppy interface but in My Computer I don't see drive A:, in DOS it just reports that drive isn't initialized and error reading partition, lights don't even blink on Gotek and I don't hear beeper

Things I've tried suggested by Chatgpt is various pin2 settings, it doesn't even makes any difference

DoYouTech, 2025-05-02:
Hello,
same troubles here.
first : 
S1 is for A:
S0 is for B: 
Then you have to put a second jumper in JC position to make it work with PC.

troydm, 2025-05-02:
@DoYouTech yeah well it works only if I boot from it (set floppy to be first in boot sequence and then boot into ms-dos setup)

it is not working in windows or when I boot into ms-dos that I have installed on separate hdd, floppy controller isn't even trying to initialize or read the drive as if it's totally ignored, is it possible that my motherboard has some kind of special floppy controller that is only activated during pc boot sequence

keirf, 2025-05-02:
JC & S1 is for both A and B. Do not use S0 with a PC. A and B are differentiated by a twist in the ribbon cable.

keirf, 2025-05-02:
If it's working in DOS then the Windows issue is a Windows issue talking to or being configured for the floppy drive controller. It's not a Gotek issue. All you need is jumpers at JC & S1, no other configuration, and the Gotek is behaving the same as a real floppy drive. You should perhaps try a real floppy drive to convince yourself that the issue is not the drive.

troydm, 2025-05-02:
@keirf yeah I have it on S1 and JC, and it works but only if I boot from floppy, it's not working in windows or ms-dos that is already installed on hdd

troydm, 2025-05-02:
@keirf it's working in dos only if I boot from floppy, it's not working in dos system that I have installed on hdd

keirf, 2025-05-02:
I would reset any cached config by deleting any FF.CFG file you may have, and pressing both buttons on the Gotek for several seconds with no USB drive inserted, until the Gotek's display changes.

keirf, 2025-05-02:
If you boot into MS-DOS, does disk changing work? Can you change disk image and the new disk image is seen by DOS?

troydm, 2025-05-02:
@keirf if I boot into floppy ms-dos setup, exit it with F3 and change image it does indeed changes to another floppy image

but when booting into ms-dos that I have installed on hdd drive A: is not working and changing floppy does nothing as it still reports error reading partition table

keirf, 2025-05-02:
Unfortunately I do not believe that this is a Gotek issue. I suggest you start by testing a real floppy drive, and ascertain a difference in behaviour between that and the Gotek.

DoYouTech, 2025-05-02:
have you checked if motherboard bios is correctly set to A: 1.44 or 2.88 ? 3 mode is supposed to be disable by default. also check if in advanced setting or if FDC controller is activated or reported for w95 (bios option).
it might be a previous BIOS setup made to prevent long delay reporting no FD or FDC.

troydm, 2025-05-02:
@DoYouTech yeah I have set it to 2.88, it was 1.44, not sure what 3 mode is as there is no such option, fdc controller is activated in bios, I don't have w95, only xp and dos on separate hdds, in xp device manager I see Standard Floppy Controller in Floppy Interfaces, however in My Computer there is no A: drive, in DOS when checking with hwinfo utility in peripheral I see floppy controller using dma2 but in IRQ assignments I see IRQ6 as free, which is weird

DoYouTech, 2025-05-02:
3 mode is a japanese floppy type, 
repporting no FDC for win95 is an option in bios to avoid troubles with early version of w95. no idea if it make the FDC unreadable in other OS but it might.
if you dont see the floppy A: in windows then it might have been removed or never detected when installing in first place. you can try refresh hardware detection in windows and see if it find the floppy now.
no idea about the Hwinfo nor the IRQ6, I havent checked that for ages with softwares.

troydm, 2025-05-07:
After few days of intensive investigations I was able to solve the issue issue thanks to chatgpt, root cause was that I was using grub4dos as bootloader for all my hdd systems, which by default disables floppy drive
https://github.com/chenall/grub4dos/issues/112

so adding this to menu.lst solved the issue
map --floppies=1

gotek drive is perfectly working from both dos and winxp

closed 2025-05-07 by troydm