keirf/flashfloppy #984: Can't get SFRKC30.AT3 into boot0 mode

sbskas, 2025-05-12:
Bought the device at amazon.

I followed the wiki and put a copper cable into the two required hole, and tried to connect to the PC with a USB-A male / male.

Device light up and shows 000 on the digital display.
However, no usb device shows up on the PC.
I think I missed something from the wiki, but can't figure what.
Any help welcome.

[image]

keirf, 2025-05-13:
If you see 000 on the display then the Gotek has not entered USB DFU mode. Are you absolutely certain you have good connection between the two jumpered pins? That should be all that's needed.

keirf, 2025-05-15:
There isn't but it's not hard to short. I do lots of Goteks just using pointy tweezers poked into the holes. The tweezers natural springiness holds them in place. They only need to be shorted for actual power on. After power on, it doesn't matter.

sbskas, 2025-05-15:
Ok, story made short, I managed to flash the board.
The holes are completely borked. Can't make a contact through. 
I had to put the tweezers on another place to short the boot0 and install the bootloader.
I followed the tracks and use two other contacts on the PCB. 
Worked first try.
However you need  to keep contact when unlocking & programming the flash else the board just disconnect.

[image]

Hope this helps.

keirf, 2025-05-15:
I'm surprised but perhaps the unlock causes a reset. That would need the BOOT0 jumper again.