keirf/flashfloppy #986: Linux Boot Disk Image

CyberExplorer, 2025-05-15:
Hello.

I have 1.44 mb boot disk img files for linux boot. I can write files with RawWrite Win to floppy and it works on machine. Also same img files work on VMWare Virtual PC. But its Ext2 linux file system. Also second image have full raw file including ramdisk for linux also. But of image cannot be loaded on flashfloppy. Its give E31. I tried different FF and IMG cfg configures. Im on latest stable 3.44

[disks.zip](https://github.com/user-attachments/files/20235477/disks.zip)

Anyone have an ideas about this ?

keirf, 2025-05-16:
The images are shortened for some reason. FlashFloppy usually works out disk image type by the file size. So try this version which has been extended to full 1440kb per image file. I did this in Linux with the command line `truncate -s 1440K disk*.img`

[disks_extended.zip](https://github.com/user-attachments/files/20239188/disks_extended.zip)

CyberExplorer, 2025-06-06:
Its worked. Thank you so much. Actually I suspect that and tried to mount original image files to Win10 VMware Virtual PC and create new image with WinImage software over Virtual PC as think about maybe this method will get me a full 1440kb file but apparently it didnt work. Next time I will use you command suggestion.

Thanks again.