keirf/flashfloppy #988: weird format challenge: SDOS (1978)

roberto314, 2025-05-24:
Hello, since i've seen FLEX .img files working with flashfloppy i would like to propose another challenge.
I am working with SDOS which are 35 Track / 18 Sector FM Images. Easy enough but Track 0 is slightly different, it starts at sector 0 and sector 5 is missing!

Output from imd2raw (convert IMD to RAW)
Source: "https://github.com/RetroFloppy/imd2raw"
Cyl 00 Hd 0 128  .................. 0  7  13 1  8  14 2  9  15 3  10 16 4  11 17 6  12 18
Cyl 00 Hd 1 128  .................. 1  7  13 2  8  14 3  9  15 4  10 16 5  11 17 6  12 18
Cyl 01 Hd 0 128  .................. 1  7  13 2  8  14 3  9  15 4  10 16 5  11 17 6  12 18

Output from: ./imdcat SDOS_System_hfe.imd -v -n
Source: "https://github.com/johnkw/dumpfloppy"
IMD 1.17: 23/04/2025 22:43:19
File generated by the HxC Floppy Emulator software v2.16.12.2

 0.0:FM-250k 18x128  0+  7+ 13+  1+  8+ 14+  2+  9+ 15+  3+ 10+ 16+  4+ 11+ 17+  6+ 12+ 18+
 0.1:FM-250k 18x128  1+  7+ 13+  2+  8+ 14+  3+  9+ 15+  4+ 10+ 16+  5+ 11+ 17+  6+ 12+ 18+
 1.0:FM-250k 18x128  1+  7+ 13+  2+  8+ 14+  3+  9+ 15+  4+ 10+ 16+  5+ 11+ 17+  6+ 12+ 18+

I have a .hfe image which boots fine but i cant get the .img files to work.
I created the .hfe images by sending the raw floppy image (equal to an .img file) via serial port to the SWTPC clone and wrote a real floppy (which booted) and than read it via greaseweazle, and converted it with HXC Floppy Emulator GUI back to .img and compared the two. The are the same.
Since it is easier to work with .img files i would like to use those.
Is there a way?

-robert

RetepV, 2025-09-07:
As far as I can tell, the .img format is just a consecutive set of sectors, without sector numbers, head numbers, track numbers, or anything else. The number of sectors, tracks, interleave, and heads expected are provided by the application. The application might guess, or the user might provide the parameters. The .img format does not specify anything for the format that it came from. Tracks to skip should also be provided by the application or the user, and there is no way to infer such a skip from the .img format.

So this is exactly a job for the .hfe format, or any other format that does include all that information (and more).

I'm also wondering why it would be easier to work with .img and .hfe formats. For applications, the .hfe format is much easier to work with, as all the parameters can be inferred from it. And for users, there is no difference. They are both simply files, right? The only difference is that the .hfe format files are larger, like 4.2MB of data for a 1.44MB format floppy disk. So .hfe format takes about 3 times more space. But well, you can still get some 15,600 HD Floppy .hfe images on a $5 64GB memory stick, so I don't really see where the problem is... ;)

Just use .hfe.

roberto314, 2025-09-08:
I agree with you, unfortunately you run into a problem when you want to BUILD a floppy image.
Now I compile or assemble a software on my computer and want to use it on the retrocomputer .
I build an image with python since the format of the filesystem is known, then I convert it to image disk (pyimagedisk) because there I have the possibility to enter the sector numbers and finally I can convert it to hfe (hxc toolbox) for the gotek.
It is a lot of work for some relatively simple problem.
Flashfloppy already supports per-track configuration and interleave so an array with the sector number arrangement in the config wouldn't be that complicated. I am however sure that this is quite the edgecase.
--Robert