keirf/flashfloppy #1000: *ERROR* 34 regardless of USB stick

number782, 2025-08-06:
GOTEK SFRKC30.AT3

flashed and working with flashfloppy fine
with new (old) pc tried to use the experimental 4.7 firmware upd

after this all USB sticks (previously working) report E34 error with LCD or *ERROR* 34 with oled screen
tried to go back to release FW
jumpered 
flashed, using dfu-util as per wiki to latest release firmware (not experimental)
dfu-util -D flashfloppy-at415-st105-3.44.dfu -a 0 -s 0x08000000:unprotect:force
dfu-util -D flashfloppy-at415-st105-3.44.dfu -a 0

FF / FlashFloppy 3.44 32kb displayed depending on Screen. 
same issue E34 or *ERROR* 34 (tried with both screens just in case)

tried formatting these sticks using Windows (FAT32 formatter) and Linux 
tried with or without FF.CFG
tried with FF.CFG in FF dir or root
tried with IMG files in subdir or root
tried multiple usb sticks

is the gotek fried or something? usb port issue? 
very strange worked with these exact usb drives, exact img files previously

keirf, 2025-08-29:
If you see Error 34 then the stick is being accessed and the filesystem being mounted. It means however that no compatible image file is found.

You say you have IMG files in the root of the USB stick. What size are they?

ccarlo64, 2025-08-29:
Hi, I'm having the same problem with a Gotek with an Artery 415. Flash updated under Linux. All good. (I've tried various versions, 3.44, 3.35, etc.) When I try to update the firmware with the two files FF_Gotek-3.* and flashfloppy-3.* etc. inside the pen drive and press the two buttons at startup, the display shows upd, prg, and then E34.
Any ideas?
When booting with nothing plugged in, I get F-F if I press ran 032, then uer 3, and then 35.
Thanks Carlo
I forgot, it writes a zero-byte file called IMAGE_A.CFG to the USB drive.

ccarlo64, 2025-08-30:
Thanks for the reply.
I don't understand which UDP files I should include. I include the ones contained in the flashdisk*.zip. Shouldn't they be those?

Carlo

number782, 2025-08-30:
> If you see Error 34 then the stick is being accessed and the filesystem being mounted. It means however that no compatible image file is found.
> 
> You say you have IMG files in the root of the USB stick. What size are they?

Hi Keir,

I know in the past if i had wrong fs type, or no images (or weird image types) I'd get E34.
Now though: Variety of images or just a few. As I said in before this issue, I was using a USB with 1.44mb imgs (windows says 1 474 560) including dos 6.22 disks from winworldpc. Also with a win98 boot disk, and a boot disk from philscomputerlab

_Each of these worked fine previously, before expermental fw was tried_
Each of these work fine when I wrote them to a real floppy disk using a Greaseweazle. 
**just the gotek would be more convienent, but it doesn't seem to like FF anymore.**

keirf, 2025-08-31:
You've probably got the wrong navigation mode configured somehow. Remove FF.CFG from the usb stick. Clear cached config in Gotek by holding both buttons after power on while no stick is inserted for a few seconds until the display changes.

number782, 2025-09-02:
> [@number782](https://github.com/number782) you've probably got the wrong navigation mode configured somehow. Remove FF.CFG from the usb stick. Clear cached config in Gotek by holding both buttons after power on while no stick is inserted for a few seconds until the display changes.

Keirf THANK YOU!
Reset worked
I was holding down both buttons as it powered up - which didn't do anything
but then i let it power up, waited, then held down the buttons until i saw "reset flash configuration"
then i did it again and pressed one of the buttons

then i powered down, powered back up with a stick with just images (no FF.CFG) and it detected my IMG file.
much appreciation for your assistance.